adafruit/circuitpython #4413: PIO: add support for the "jmp pin"

jepler, 2021-03-15:
The PIO module can support a conditional jump based on a GPIO pin.  However, the pin to use can't be specified in CircuitPython. Let's add it!

daveythacher, 2021-04-04:
Related files:
https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/bindings/rp2pio/StateMachine.c
https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/rp2pio/StateMachine.c

NathanY3G, 2021-07-22:
Hi. If there are no objections I would like to take this issue on!

jepler, 2021-07-22:
@NathanY3G thank you for offering! I've assigned the issue to you. Let us know how we can help you get up to speed working on the CircuitPython core—our [Discord](https://adafru.it/discord) is a great place to find help as well, just drop in to #circuitpython-dev and ask a question.

NathanY3G, 2021-07-22:
Cheers Jeff :-)

dmcomm, 2021-07-22:
Looking forward to it :)

NathanY3G, 2021-07-23:
I have created a pull-request and would be grateful for some feedback, especially since this is my first contribution to CircuitPython: https://github.com/adafruit/circuitpython/pull/5042

Testing: this change was manually tested on a Feather RP2040 and a RPi Pico by using the attached test.

[jmp_pin_interactive_test.txt](https://github.com/adafruit/circuitpython/files/6867232/jmp_pin_interactive_test.txt)

closed 2021-07-23 by microDev1