adafruit/circuitpython #4414: Fast copy from ulab to displayio.Bitmap

dglaude, 2021-03-15:
Playing with ulab, I found it was costly to copy from an array to a display.bitmap.

ulab can build a buffer: ulab.array.tobytes

Here is the current fastest way I was able to copy:
```
image_bitmap = displayio.Bitmap(24, 32, number_of_colors)
inta=ulab.array((npframe-min_t)*factor,dtype=ulab.int8)
index = 0
for h in range(24):
    for w in range(32):
        image_bitmap[h, w] = inta[index] 
        index+=1

```

v923z, 2021-03-15:
Basically, what would be needed is a way of passing the bytes of an ndarray directly into the bitmap. One option would be the `tobytes` array method, if `displayio` could accept a buffer. https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html?highlight=tobytes#tobytes

kmatch98, 2021-03-15:
I'm interested in helping out on this and stretching my skills a bit to try to understand better how the bitmap structure is formed and modified.  Sorry I didn't chime in during today's CircuitPython call, but I'm not very well versed in the terminology y'all use, so I have to do some digging to be partially intelligible when discussing this.  Here goes:

As best I understand, to save memory the bitmap structure uses the minimum number of bits necessary to hold the maximum palette indices (`value_count` in put parameter to `displayio_bitmap_make_new()`.  For that reason, each pixel could be represented by somewhere between 1 and 32 bits (either 1-8, 16 or 32 bits).    

So, the `ndarray` will somehow need to know the `bits-per-value` of the bitmap, the `width`, `height`, and `stride`.

In looking at the 
[ulab documentation](https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html?highlight=tobytes#the-ndinfo-function), it looks like ndarrays can be decoded as 8, 16 or 32 bits.  (I'm uncertain if < 8 bits can be handled directly in ndarray.)

I've heard y'all talk about using "buffer protocols" to handle this, but I'm not very familiar with this, and @v923z you mention adding a `buffer` keyword into the bitmap constructor, and @jepler mentioned `mp_get_buffer` on the CircuitPython call today.  In a brief reading about buffers, it seems like it's a chunk of memory that you can access in different ways.  Now I see why you are suggesting this.  Can y'all point to an example where a similar thing like this is done?

v923z, 2021-03-16:
> So, the `ndarray` will somehow need to know the `bits-per-value` of the bitmap, the `width`, `height`, and `stride`.

No, that should definitely be done on the bitmap's side. 

> 
> In looking at the
> [ulab documentation](https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html?highlight=tobytes#the-ndinfo-function), it looks like ndarrays can be decoded as 8, 16 or 32 bits. (I'm uncertain if < 8 bits can be handled directly in ndarray.)

Bits can't be stored even in `numpy`. Booleans are just `uint8` with a special flag. 

> I've heard y'all talk about using "buffer protocols" to handle this, but I'm not very familiar with this, and @v923z you mention adding a `buffer` keyword into the bitmap constructor, and @jepler [...]

I can only speak for `ulab`, but here are two examples: https://github.com/v923z/micropython-ulab/blob/714a3b872747332de9600fda4be07bbcc5f99626/code/ulab_create.c#L693-L705, and the reverse https://github.com/v923z/micropython-ulab/blob/714a3b872747332de9600fda4be07bbcc5f99626/code/ndarray.c#L1595-L1606

closed 2021-03-17 by jepler