adafruit/circuitpython #4417: Can't disable mass storage for the Pimoroni tiny 2040

ArielWolle, 2021-03-15:
Error when building 

**Code/REPL**

adding the line: CIRCUITPY_USB_MSC = 0
to /ports/raspberrypi/boards/pimoroni_tiny2040/mpconfigboard.mk results in a build error 

**Behavior**

[image]

**Description**

Trying to disable mass storage as i need to write to the tiny 2040 using the storage module but dont want to eject every time i plug it in. Being used as an HID device so needs to be plugged into data and power of usb not just power.

Removing [this](url) line resolves the issue.

ladyada, 2021-03-16:
@ArielWolle wanna try submitting a PR where the line is behind an #ifdef or #ifnef (the url isnt working)

dhalbert, 2021-03-16:
We thought this was fixed by #4283, which was merged, so it should be in the latest source base. Could you paste the source URL again?

ArielWolle, 2021-03-16:
i a git cloned staright from the main branch url yesterday i will try again right now

ArielWolle, 2021-03-16:
Just tried rebuilding again using "git clone https://github.com/adafruit/circuitpython.git" and same issue works as soon as I take that line out

deshipu, 2021-03-16:
It's possible it worked with my test board because I had ``CIRCUITPY_STORAGE`` disabled.

@ArielWolle which line exactly? Your link point to nowhere.

ArielWolle, 2021-03-16:
I added the line 
CIRCUITPY_USB_MSC = 0
to the bottom of:
https://github.com/adafruit/circuitpython/blob/1be5ca7881a8aa8d32a783bc6a94994886463c30/ports/raspberrypi/boards/pimoroni_tiny2040/mpconfigboard.mk

miclooking, 2021-03-22:
> 
> 
> I added the line
> CIRCUITPY_USB_MSC = 0
> to the bottom of:
> https://github.com/adafruit/circuitpython/blob/1be5ca7881a8aa8d32a783bc6a94994886463c30/ports/raspberrypi/boards/pimoroni_tiny2040/mpconfigboard.mk
> 
> This is what my mpconfigboard.mk board looks like
> 
> [image]

I get the same error when adding CIRCUITPY_USB_MSC = 0 in mpconfigboard.mk for Raspberry Pi Pico.
Is there any solution for this?
Thanks.

PR #4462 (2021-03-22): storage: Correct when we check for USB mounts
Closes #4417

I only compile-tested this for feather rp2040 with 
```diff
diff --git a/ports/raspberrypi/boards/adafruit_feather_rp2040/mpconfigboard.mk b/ports/raspberrypi/boards/adafruit_feather_rp2040/mpconfigboard.mk
index f4106b94a..f891c3668 100644
--- a/ports/raspberrypi/boards/adafruit_feather_rp2040/mpconfigboard.mk
+++ b/ports/raspberrypi/boards/adafruit_feather_rp2040/mpconfigboard.mk
@@ -7,3 +7,5 @@ CHIP_VARIANT = RP2040
 CHIP_FAMILY = rp2
 
 INTERNAL_FLASH_FILESYSTEM = 1
+
+CIRCUITPY_USB_MSC = 0
```

@arielwolle @miclooking please test [... cut]
commit 5a0e9945 (2021-03-22): storage: Correct when we check for USB mounts

Closes #4417
closed 2021-03-22 by tannewt