adafruit/circuitpython #4420: Sockets do not work reliable on a ESP32-S2 / carried over from "uPy socket vs cPy socket need help" topic in the forum on request of Dan

ThomasAtBBTF, 2021-03-16:
**Firmware**

```python
With 6.2.0-beta.2 and 6.2.0-beta.3 on a MagTag
```

**Code/REPL**

```python

import sys
import ipaddress
import time
import wifi
import socketpool
import adafruit_requests
import ssl
import board
import busio
import json

MYSERVER = "HTTP/1.1 200 OK\nServer: WebSocket Server\nContent-Type: text/html\n"
MYHTML_LEN = "Content-Length: {}\n\n"
MYHTML = """
<!DOCTYPE html>
<html>
	<head> 
		<style>
			title {
				text-align: left;
			}
			body {
				background-color: #FFC926;
			}
			h1 { 
				background-color: green;
				color: white;
			}
			th {
				text-align: left; 
				color: blue;
				font-family: sans-serif;
			}
			td {
				text-align: left; 
				color: black;
				font-family: sans-serif;
			}
		   input {
				background-color: #C2DFFF;
				}
		   select{
				background-color: #C2DFFF;
				}
	</style>
		<title>ESP32 Webserver Demo V 0.1</title> 
	</head>
	<body>
		<h1 style="text-align: center;">Demo</h1>
		<form ID="form" name="form">
			<table>
				<tbody>
					<tr> <th>Mode:</th>
					<td> 
						<select ID="MYMODE">
							<option selected>ON</option>
							<option >OFF</option>
							<option >MANUAL</option>
						</select>
					</td>                                                                                         
[... 498 lines cut ...]

This project shows how a controller running CircuitPython can send a website to a browser to create an interactive user interface with select objects, edit fields and buttons.
The controller can fill in data into these objects and the JavaScript in the website can handle the data from the controller.
Also, the JavaScript can tell the controller about the state of the website (focus location).
Also, the JavaScript can tell the controller about the UI events like selection in the "dropdown" or clicking buttons.

All this is implemented by periodical get requests in a JavaScript-Timer and Events from the UI-Elements.
To transfer the data and to receive data from the controller the Java XMLHttpRequest class is used.

I am observing the following problems:
1. when sending the website the send_line procedure often sends only some bytes or I get an EAGAIN error.
The code in send_line is handling the problem and typically the full website is transferred to the browser.

2. "after while" requests are not arriving anymore at the sockets...
That request are generated and are "timing-out" can be seen in the debug screen in Chrome (which can be activated by pressing F12)

Sorry, but I can not strip the code much more as multiple components are at play in order to reproduce the problem which are:
1. a MagTag with CP-code
2. a PC with Windows 10 
3. Chrome running on this PC
4. a WiFi Network which connects the MagTag and the PC

**Additional Info**

I will try to upload a commented log file from my controller connected to a console with also some print screen as a PDF.

ThomasAtBBTF, 2021-03-16:
https://www.dropbox.com/s/sir61s4bqkulpnp/miniWebServer%20log.pdf?dl=0

microDev1, 2021-03-16:
Carried over from forum: [uPy socket vs cPy socket need help](https://forums.adafruit.com/viewtopic.php?f=60&t=176679).

tannewt, 2021-03-16:
It's worth trying a latest build. I believe there have been changes since beta.3 that may have improved things.

ThomasAtBBTF, 2021-03-16:
This code works much better now.
[magtagserver.zip](https://github.com/adafruit/circuitpython/files/6151215/magtagserver.zip)
Closing the sockets after use did the trick.
This seemed not to be a problem so much in MicroPython.

dhalbert, 2021-04-07:
@ThomasAtBBTF So the essence of your fix is to close the sockets? Maybe we need to document that? Should we keep this issue open and perhaps change the title?

ThomasAtBBTF, 2021-04-07:
@dhalbert 

Yes, the need to close the sockets certainly should be documented. 
As this "solves" my problem 2.

(Anyway, without that, I suppose there is a chance to "leak memory".)

But my observation that send often does not "send everything" remains.  (Problem 1) 
Here my "send_line" code:

    def send_line(self, l, byte_line):
        need_bytes = len(byte_line)
        sum_bytes = 0
        retries = 0
        while True:
            try:
                line_bytes = self.socket.send(byte_line)
                sum_bytes += line_bytes
                # preparing some of the bytes which are send for debug output
                s = byte_line.decode("UTF-8")
                s = s[:30]
                if sum_bytes == need_bytes:
                    # the tansfer succeded to send all bytes
                    c = " "
                    if l > 0:
                        print("\r\nl:", c, retries, l, sum_bytes, need_bytes, s, end="")
                    return sum_bytes  # return because we are done!
                else:
                    retries += 1
                    c = "?"
                    s = s[:line_bytes]  # show the bytes which were successfully transfered
                    if l > 0:
                        print("\r\nl:", c, retries, l, sum_bytes, need_bytes, s, end="")
                    byte_line = byte_line[line_bytes:]  # slice away the already successful transfered bytes
                    continue  # try again with the remaining bytes
            except OSError as e:
                if e.errno == 11:  # EAGAIN  here no bytes have been transfered
                    print("\r\nEAGAIN error: no bytes transfered, will try again!", end="")
                    time.sleep(0.1)  # wait a while.... ? neccessary ?
                    continue
                elif e.errno == 104:  # ECONNRESET  here the other side is not responding anymore!
                    self.valid = False  # ! flag this connection / client as not valid anymore
                    print("\r\nECONNRESET error: no bytes transfered, will abort! ", self.socket, end="")
                    return -1
                else:
                    print("\r\nsend_line Error:", e.errno, str(e), end="")  # another error (not observed so far!)
                    return -1

I would think, that this should be handled "inside" the socket until retries get's too high.
With this code above I am eventually able to send everything, assuming the WiFi connection does not break.

closed 2021-04-07 by ThomasAtBBTF