adafruit/circuitpython #4431: Memory/symbol table corruption relating to audio or pwm on Pi Pico

kevinjwalters, 2021-03-18:
Running this on `6.2.0-beta.3-11-g7970c882a on 2021-03-04` on a Pi Pico

```
import time
import math
import random
import struct

import board
import pwmio
import digitalio
from audiopwmio import PWMAudioOut as AudioOut
from audiocore import WaveFile
import rp2pio
import adafruit_pioasm

AUDIO_PIN_L = board.GP18
AUDIO_PIN_R = board.GP19
audio_out = AudioOut(AUDIO_PIN_L, right_channel=AUDIO_PIN_R)
stereo_file1 = open("scanner-left-16k.wav", "rb")
stereo_file2 = open("scanner-right-16k.wav", "rb")

buffer1 = bytearray(80 * 1024)
buffer2 = bytearray(80 * 1024)
sound_sample1 = WaveFile(stereo_file1, buffer1)
sound_sample2 = WaveFile(stereo_file2, buffer2)

def start_sound(sample):
    if sample is not None:
        audio_out.play(sample)

def wait_sound():
    while audio_out.playing:
        pass
        

print("SLEEP 12")
time.sleep(12)

pwm_gp4 = pwmio.PWMOut(board.GP4, frequency=7000, duty_cycle=0)
pwm_gp5 = pwmio.PWMOut(board.GP5, frequency=7000, duty_cycle=0)
pwm_gp6 = pwmio.PWMOut(board.GP6, frequency=7000, duty_cycle=0)
pwm_gp7 = pwmio.PWMOut(board.GP7, frequency=7000, duty_cycle=0)
pwm_gp8 = pwmio.PWMOut(board.GP8, frequency=7000, duty_cycle=0)
pwm_gp9 = pwmio.PWMOut(board.GP9, frequency=7000, duty_cycle=0)
pwm_gp10 = pwmio.PWMOut(board.GP10, frequency=7000, duty_cycle=0)
pwm_gp11 = pwmio.PWMOut(board.GP11, frequency=7000, duty_cycle=0)
pwm_gp12 = pwmio.PWMOut(board.GP12, frequency=7000, duty_cycle=0)
pwm_gp13 = pwmio.PWMOut(board.GP13, frequency=7000, duty_cycle=0)
pwm_gp14 = pwmio.PWMOut(board.GP14, frequency=7000, duty_cycle=0)
pwm_gp15 = pwmio.PWMOut(board.GP15, frequency=7000, duty_cycle=0)
pwm_gp16 = pwmio.PWMOut(board.GP16, frequency=7000, duty_cycle=0)
pwm_gp17 = pwmio.PWMOut(board.GP17, frequency=7000, duty_cycle=0)

print("CHECKING SUB", globals()["wait_sound"])

while True:
    print("PLAY 1")
    start_sound(sound_sample1)
    print("WAIT 1")
    wait_sound()
    time.sleep(1)

    print("PLAY 2")
    start_sound(sound_sample2)
    print("WAIT 2")
    wait_sound()
    time.sleep(1)
```

Gives this output:

```
Adafruit CircuitPython 6.2.0-beta.3-11-g7970c882a on 2021-03-04; Raspberry Pi Pico with rp2040
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

code.py output:
SLEEP 12
CHECKING SUB <function wait_sound at 0x20006030>
PLAY 1
WAIT 1
Traceback (most recent call last):
  File "code.py", line 60, in <module>
NameError: name 'wait_sound' is not defined

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

Commenting out `pwm_gp4` line magically fixes the problem but it's very unclear what's the actual cause of this.

The buffer sizes are unusually high. I was making them bigger than the wav file to try and fit the whole thing in memory.

This could relate to #4208. The nrf code has issues around buffers, see #3030

dhalbert, 2021-03-18:
These buffers are > 65535 bytes. I wonder if there's a 16-bit limit somewhere.

kevinjwalters, 2021-03-23:
I can also hang the Pi Pico with code mentioned in https://github.com/adafruit/circuitpython/issues/4208#issuecomment-804296887

eigenein, 2021-04-25:
I'm experiencing the same with `pulseio.PulseIn`:

```python
from board import GP16
from pulseio import PulseIn

pulse_in = PulseIn(GP16, maxlen=1)

for j in range(5):
    print(f"Run #{j}:")
    pulse_in.clear()
    while not pulse_in:
        pass
    print([pulse_in[j] for j in range(len(pulse_in))])
```

While signal is coming to `GP16`, I'm getting a similar output:

```text
Run #0:
[141, 141]
Run #1:
[141, 141]
Run #2:
[141, 141]
Run #3:
[141]
Run #4:
Traceback (most recent call last):
  File "code.py", line 8, in <module>
NameError: name 'pulse_in' is not defined
```

It occurs on a pretty random iteration.

dhalbert, 2021-08-12:
I was seeing crashes with explicitly-specified buffers as well, and when they were only of size 2048. This may be fixed by #5079, but I need to re-test

dhalbert, 2021-08-17:
This no longer crashes, but supplying a buffer >1025 raises the exception `RuntimeError: Internal audio buffer too small`.

(1025 only works I think because these are 16-bit samples, so 1024+1 works because the extra byte is ignored, but 1025 does not work).

I'm going to close this and open a new issue about that, since it's not fatal.

By the way, I tried this with all the PWM pins, and with them commented out, and did not heard a difference.

closed 2021-08-17 by dhalbert