adafruit/circuitpython #4433: Most appropriate way to change device name?

infovore, 2021-03-18:
I've been prodding this library as an alternative to some other code I have, and it's super-impressive. One question: what's the most appropriate way to change the auto-generated device name?

In my MIDI Monitor tool, I get this:

[image]

and I've had good success with other boards on changing the detected devicename - which I appreciate might be a USB setting rather than a MIDI-specific setting. I'm new to CP, so am imagining this is a `settings.py` or `boot.py` issue, but don't quite know where to begin.

dhalbert, 2021-03-18:
transferred and re-titled

kevinjwalters, 2021-03-18:
@infovore How do two boards show up in that tool?

samkusnetz, 2021-12-05:
hello. i'm a relatively new user of circuitpython, recently crossing over from my long, long time with arduinos. i would enthusiastically appreciate this feature!

dhalbert, 2023-10-28:
Superseded by #4191.

closed 2023-10-28 by dhalbert