adafruit/circuitpython #4440: REPL slow to connect on nRF52

dhalbert, 2021-03-19:
@hathach I have noticed that the REPL is slow to appear in CircuitPython on nRF52, after connecting via a terminal program. I have seen this in Linux and after power-up or hard rest. This has been true for a while now. I will try to characterize this better but am opening this issue at least as a placeholder.

hathach, 2021-03-19:
is it always slow or just start to be slow recently. I will try to narrow down the scope to analyze the issue.

dhalbert, 2021-03-20:
@hathach I did a bisect and narrowed this down to:

commit 5351a93c70fcacc50a154edac0a6d6d1028296e0
Author: hathach <thach@tinyusb.org>
Date:   Mon Nov 23 13:39:14 2020 +0700

    update tinyusb to fix cdc connection race
    
    issue is fixed in https://github.com/hathach/tinyusb/pull/557

Before this commit, connecting to the REPL on a Feather nRF52840 responds immediately. Afterwards, it takes about five seconds after connecting to print the initial prompt. I looked in `/var/log/syslog` and nothing is printed during that time, so it seems all internal.

hathach, 2021-03-20:
thanks @dhalbert I will look at this next week, there is also a relevant fix for dcd of nrf52 OUT transfer that could cause drop of packet recently, will see if that also fix this.  https://github.com/hathach/tinyusb/pull/733

hathach, 2021-04-08:
@dhalbert I couldn't reproduce the issue with latest 0539b88a6bd4237ef02ea35cd5e97f638fd132df, the REPL connected immediately, i am using minicom on ubuntu 20.04. How is your testing,

dhalbert, 2021-04-08:
I am still seeing this on Linux with Ubuntu 20.04. I have an Intel-based motherboard. (I think yours may be AMD?) Seen on both a Feather Sense and a CPB. There is no `code.py` on the board, so it goes straight to the REPL.
Here's a Beagle trace:
[nrf-repl-5-second-delay.tdc.zip](https://github.com/adafruit/circuitpython/files/6280580/nrf-repl-5-second-delay.tdc.zip)
Steps:
1. Start trace
2. plug in board
3. wait a few seconds for enumeration to finish; confirmed by watching `/var/log/syslog`
4. connect via picocom (issue seen with minicom also)
5. it takes 5 seconds for `Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.` prompt to appear. During and after that time, there is no extra output in `/var/log/syslog`.

dhalbert, 2021-04-08:
I tried this on Windows also, with Tera Term, and it never actually seems to connect, which I haven't seen before. I could get a trace there too, if you want, but you have a Windows machine also.

hathach, 2021-04-09:
Thanks @dhalbert I will pull out my windows machine to test with

PR #4900 (2021-06-21): update tinyusb to fix nrf control race condition
fix race condition with control since TASKS_EP0RCVOUT also require EsyDMA. Should fix #4894 and fix #4440 .
closed 2021-06-21 by dhalbert