adafruit/circuitpython #4444: REPL can get wedged on mimxrt

jepler, 2021-03-19:
Be careful, following these instructions can leave you with a board in a difficult to recover state!

1. Put this in code.py:
  ```python
print("hello world")
while True:
    pass
```
2. reset the board
3. connect via usb-cdc to repl
4. hit "A" then "ctrl-c"

Normally, you'd expect the ctrl-c to interrupt.  However, it doesn't.  If "ctrl-c" is the _FIRST_ thing typed through usb-cdc, then it probably works.  But typing anything else first leaves it non-working. My Linux terminal program announces that it can't send characters through the serial link anymore.

@ladyada and I both ran into a form of this, me on the mimxrt1010-evk and her on the metro m7 prototype.

We're assuming it's mimxrt10xx only but didn't test other micros.

jepler, 2021-03-22:
@dhalbert this was my interpretation of what Limor was seeing, but it could be something else..

DavePutz, 2021-04-09:
This looks like it started happening at commit accc454 in CP (perhaps due to picking up new tinyusb commits, where it looks to have started with commit 1e2c554). I haven't yet identified the  code causing the issue, although it appears that the USB interrupt handler stops getting called after the first character of input.

jepler, 2023-03-27:
This is still happening. However, the new "UART low level break" code does make it snap out of this situation.

What's different between metro m7 and other boards is that this happens after just 1 byte! on other boards it took more pending UART data (like, say, 64 bytes) so you wouldn't hit it as easily/often.

see 
 * https://github.com/adafruit/circuitpython/pull/7227
 * https://github.com/adafruit/circuitpython/issues/7183

tannewt, 2023-03-28:
This feels a little bit like a watermarking issue where the interrupt isn't raised until N bytes. However, I suspect it is more likely a caching error where the USB peripheral has updated the buffer but the CPU sees an older version. I thought I'd fixed this by aligning the CDC buffer to 32 bytes (a cache line) but may have missed a case.