adafruit/circuitpython #4448: QT PY: no yellow status LED on boot

Neradoc, 2021-03-20:
**Firmware**

```python
Adafruit CircuitPython 6.2.0-beta.2 on 2021-03-01; Raspberry Pi Pico with rp2040
```

**Code/REPL**

```python
import busio, bitbangio
i2c = bitbangio.I2C(board.GP1, board.GP0)
```

**Behavior**

```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TimeoutError: Clock stretch too long
```

**Description**

- Error while using i2c...
- Only happens when...
- might be related to #4291...

**Additional Info**

Removing [this](url) line resolves the issue.

dhalbert, 2021-03-21:
Related to #4451.

kattni, 2021-05-05:
The ItsyBitsy RP2040 also does not show yellow on start up.

kattni, 2021-05-20:
This is resolved by #4743. It can be closed when that PR is merged. Tested on both Itsy and QT Py.

closed 2021-05-21 by Neradoc