adafruit/circuitpython #4450: Suspicion of wrong dirty rectangle tracking for transposed tilegrids

jepler, 2021-03-20:
I haven't distilled this down into a test program, but I had a tilegrid with 8x5-pixel tiles, sized 4x23, and transposed. (so it ends up being a wide and short tilegrid)

I found that after the initial draw, only a narrow strip at the left of the tilegrid would be updated when I changed the tiles.

Here are the relevant snippets.  Running with recent main branch on a feather rp2040.
```
font = displayio.Bitmap(8, 5*256, 2) 
with open("/font5x8.bin", "rb") as f:
    f.read(2)
    bitmaptools.readinto(font, f, bits_per_pixel=1)
    
palette = displayio.Palette(2)
palette[1] = 0xff0000
label = displayio.TileGrid(font, pixel_shader=palette, height=23, width=4, tile_width=8, tile_height=5)
label.transpose_xy = True
label.x = 5
group = displayio.Group()
group.append(label)
DISPLAY.show(group)

...
# populate rows with strings to show
    DISPLAY.auto_refresh=False 
    for i, r in enumerate(rows):
        for j, c in enumerate(r):
            #print((i,j, c))
            label[i,j] = ord(c)
        for j in range(j+1, 23):
            label[j,i] = 32
    DISPLAY.auto_refresh=True
```

jepler, 2021-03-20:
.. I'm working around it by modifying the palette, which causes the whole tilegrid to redraw ..

kmatch98, 2021-03-21:
I replicated your issue when writing into a transposed bitmap. I get strange behavior that depends upon the bitmap size.  I’ll dig into this over the next couple of days. 

Mainly I need to build my debugging skills on the CP core side and this is a good trial with watching the dirty rectangle updates.

jepler, 2021-03-21:
Excited to see you stepping up to look into this @kmatch98 -- thanks in advance!

tannewt, 2021-03-22:
I think this was fixed by #4464

closed 2021-03-22 by tannewt