adafruit/circuitpython #4451: QT Py board.NEOPIXEL_POWER is always in use

dhalbert, 2021-03-21:
On the Qt Py,  `board.NEOPIXEL_POWER` cannot be controlled, by, say, doing `digitalio.DigitalInOut(board.NEOPIXEL_POWER)`. The pin is always in use, because it is set to high in `board.c` and also set to never reset. But if the `never_reset` is removed, then the pin does get reset, and the status NeoPixel goes out. So it needs to be special-cased in some way.

dhalbert, 2021-03-21:
Related to #4448.

tannewt, 2021-03-22:
Ya, I think we need to add a mechanic for board power management. It'd be able share the pin with user code like neopixel and dotstar already do.

Neradoc, 2021-05-21:
Fixed by #4743

closed 2021-05-21 by dhalbert

dclaar, 2022-01-25:
This still seems broken on https://learn.adafruit.com/adafruit-qt-py-esp32-s2 at 7.1.1.
Should I open a new issue for that?

```
>>> import digitalio
>>> import board
>>> np_power = digitalio.DigitalInOut(board.NEOPIXEL_POWER)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: NEOPIXEL_POWER in use
```
boot_out.txt:
```
Adafruit CircuitPython 7.1.1 on 2022-01-14; Adafruit QT Py ESP32S2 with ESP32S2
Board ID:adafruit_qtpy_esp32s2
```

anecdata, 2022-01-25:
Seems to have been fixed on the `main` branch:
```py
Adafruit CircuitPython 7.2.0-alpha.1 on 2021-12-29; Adafruit QT Py ESP32S2 with ESP32S2
>>> import digitalio
>>> import board
>>> np_power = digitalio.DigitalInOut(board.NEOPIXEL_POWER)
>>> 
```

dclaar, 2022-01-25:
Yes, thanks!