adafruit/circuitpython #4453: PWMAudioOut audio output uses last pins set even if code is changed on nRF52840

kevinjwalters, 2021-03-21:
I was just playing some audio on a Feather nRF52840 and changed the code from stereo to mono and I could still see the audio was active on the `right_channel`. I could see on the serial console that it was doing Auto-reload but I verified `code.py` had actually changed. I did a `microcontroller.reset()` and then it was actually using the pins I defined. I'm just watching audio on LEDs so not sure if samples will then play on both channels but there was certainly activity on right_channel when there should not have been.

This is on 6.1.0.

```
import board
from audiocore import WaveFile
try:
    from audioio import AudioOut
except ImportError:
    from audiopwmio import PWMAudioOut as AudioOut

stereo=True   ### change this to False
AUDIO_PIN_L = board.A0
AUDIO_PIN_R = board.A1
if stereo:
    audio_out = AudioOut(AUDIO_PIN_L)
                             right_channel=AUDIO_PIN_R)
else:
    audio_out = AudioOut(AUDIO_PIN_L)
```

You need to actually play a sample to get the PWM outputs to light up.

kevinjwalters, 2021-03-21:
BTW the PWM does not continue after a sample is playing resulting in clicks. I think this differs to the RP2040 implementation which carries on PWMing at `quiescent_value`. Is that a bug? I noticed it today on 6.1.0 and wayback when writing and playing [Adafruit Learn: Quick Draw Duo](https://learn.adafruit.com/circuit-playground-bluefruit-quick-draw-duo/quick-draw-duo?embeds=allow#example-video-3051524-8). You can hear the clicks in the video as thuds at the end of the exaggerated richochet.

dhalbert, 2026-11-13:
Claude wrote this:

This no longer happens. Objects are now freed by their finalizers when the VM ends. On an ItsyBitsy nRF52840 running 11.0.0-alpha.3, a stereo `PWMAudioOut` on A0/A1 left playing across a soft reload, followed by a mono `PWMAudioOut` on A0, leaves only A0 connected to the PWM; the right-channel output is disconnected.

closed 2026-11-13 by dhalbert