adafruit/circuitpython #4461: raspberrypi: rare hard locks when writing flash while RGBMatrix is active

jepler, 2021-03-22:
I spent much of the weekend developing a project on the Feather RP2040 that uses RGBMatrix.

Occasionally (1 in 20?) times when I would write out code.py, the board would freeze (USB stopped, matrix scanning stopped).  A hard reset cured it, and I never lost my CIRCUITPY contents.

@ladyada also saw this at least once while she was testing the code, but we decided at the time to brush it off.  Unfortunately, it's real and it's still there.

daveythacher, 2021-04-03:
Can you post code?

jepler, 2021-04-03:
There's no code per se, because it locks up during disk activity and/or soft reload.  I assume that the code from the guide will set up the conditions for eventual failure:
```python
displayio.release_displays()
matrix = rgbmatrix.RGBMatrix(
    width=64, bit_depth=4,
    rgb_pins=[board.D6, board.D5, board.D9, board.D11, board.D10, board.D12],
    addr_pins=[board.D25, board.D24, board.A3, board.A2],
    clock_pin=board.D13, latch_pin=board.D0, output_enable_pin=board.D1)
display = framebufferio.FramebufferDisplay(matrix)
```

In my specific case, I would be repeatedly editing & saving a ~8kB (240 line) code file using vim on a raspberry pi 4 as the host computer.