adafruit/circuitpython #4466: i2c default clock speed for Pi Pico (RP2040) is 90.9kHz

kevinjwalters, 2021-03-23:
I've not had any particular i2c issues but looking at #4082 the clock speed looked wrong. Here's a view from Ikalogic SQ25 of a (Maker) Pi Pico talking to a generic SSD1306-driven screen and using the change contrast command to get something to capture (at 25MHz).

[image]

This has the 10k pullups on SSD1306 and whatever the RP2040 might be doing.

I took those 3 measurements and 3 others to get mean of 90.8827kHz. That's suspiciously close to 1000/11 = 90.9090.

Would be useful if someone could confirm i2c speed on Pi Pico and perhaps another established processor.

```
Adafruit CircuitPython 6.2.0-beta.4 on 2021-03-18; Raspberry Pi Pico with rp2040
>>> import board
>>> import busio
>>> i2c = busio.I2C(scl=board.GP1, sda=board.GP0)
>>> i2c.try_lock()
True
>>> buffer = bytearray(2)
>>> buffer[0] = 0x81  ### set contrast command
>>> buffer[1] = 0x80
>>> i2c.writeto(60, buffer)
```

Same result with `frequency=100_000`.

Oddly, `frequency=110_000` gives actual 111.193kHz, `frequency=400_000` gives actual 250.000 kHz

[image]

kevinjwalters, 2021-03-23:
I tried some 4 byte writes as I'm pretty sure I've read a discussion (perhaps with @Gadgetoid) about the special case bitbang code kicking in at `<= X` where `X` may equal 2.

The trace looks very different for 4 byte write. Here's first part. It's sped up a bit to 265kHz but it's still not the requested 400kHz.

[image]

Gadgetoid, 2021-03-23:
@kevinjwalters yes currently any write <= 2 bytes will go through a bitbangio instance:

https://github.com/adafruit/circuitpython/blob/98425c9a9d594d05249bcf4e8edf47075d613111/ports/raspberrypi/common-hal/busio/I2C.c#L159-L161

I've got an outstanding PR to change this to `== 0` since rp2040 can handle 1 and 2 byte writes fine but will not do a 0-byte address-only write (used to check for the address ACK when probing the bus). I think this was the result of some misunderstanding or very early SDK bug.

dhalbert, 2021-03-23:
@kevinjwalters The 2 bytes-or-fewer writes are done using bitbangio and I would expect their timing to be approximate. The other off-timing may be due to clock divisor granularity.

kevinjwalters, 2021-03-23:
FYI, my PyPortal (SAMD51 / M4) talking to its ADT7410 running `6.1.0` matches the two traces @caternuson put in the other ticket at 94.697 kHz.

kevinjwalters, 2021-03-24:
@dhalbert 786k is the new 400k.

```
Adafruit CircuitPython 6.2.0-beta.4 on 2021-03-18; Raspberry Pi Pico with rp2040
>>> import board, busio
>>> i2c = busio.I2C(scl=board.GP1, sda=board.GP0, frequency=786_000)
>>> big_buf = bytearray([0x81, 0x46, 0x81, 0x55])
>>> i2c.try_lock()
True
>>> i2c.writeto(60, big_buf)
```

From three runs measuring a chunk of clock pulses

```
Average Frequency = 399.436 kHz
Average Frequency = 401.070 kHz
Average Frequency = 399.787 kHz
```

Does this mean it's a bug in https://github.com/raspberrypi/pico-sdk ?

dhalbert, 2021-03-24:
The speeds you ask for may not be precisely available based on the way the peripheral clock is generated. It's usually generated by dividing a higher-speed clock. I don't think it's necessarily a bug; you'd need to study the clock setting in the I2C driver.

This is common: for instance, when we have high SPI speeds on other chips, they are very approximate because the divisors are quite small.

dhalbert, 2021-03-24:
So based on your testing of 4-byte writes this issue title could be revised.

kevinjwalters, 2021-03-24:
I graphed the data

[image]

Would be good is someone else could check it in case my SQ25 is misbehaving but that seems unlikely.

dhalbert, 2021-03-25:
We just pass the frequency on to the `pico-sdk`. Its frequency-setting code is here:
https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_i2c/i2c.c#L63=L90.

kevinjwalters, 2021-03-25:
That's what I thought. I put in https://github.com/raspberrypi/pico-sdk/issues/283 for this. It's curious the bitbangio code is also slow. I'll close this when it's acknowledged by RPi folk.

kevinjwalters, 2021-03-25:
This is the RP2040 doing clock stretching apparently but will be interesting to see if it's doing it in appropriate places for fast mode.

closed 2021-03-25 by kevinjwalters