adafruit/circuitpython #4467: Add native audio synthesis

tannewt, 2021-03-23:
Right now we can only playback RawSamples, WAV and MP3. It'd be nice to add a `synthio` module that can generate audio based on settings set from Python. For buffering changes, it could also take in MIDI commands and apply them over time.

The Teensy Audio library is a really neat Arduino library we can reference: https://github.com/PaulStoffregen/Audio

It has these synth modules:
- wavetable
- drum
- string
- sine
- sine_hires
- sine_fm
- waveform
- waveformMod
- pwm
- tonesweep
- dc
- noise
- pink

These effects:
- fade
- chorus
- flange
- reverb
- freeverb
- freeverbs
- envelope
- multiply
- rectify
- delay
- delayExt
- bitcrusher
- midside
- waveshape
- granular
- combine

These filters:
- biquad
- fir
- filter

These analyzers:
- peak
- rms
- fft256
- fft1024
- tone
- notefreq
- print

tyomitch, 2021-04-08:
API-related comments, originally from #4447: 

The synthesiser can implement the AudioSample protocol, but in addition to that, we'd like to be able not to pass it a `buffer` upfront, but to call a `write` method instead that queues data up. That way we'd be able to pipe MIDI from BLE or USB directly into this object.

OTOH,
* that would make looping (or even passing the same AudioSample to `play()` repeatedly) impossible;
* the protocol for `audiosample_reset_buffer_fun` has no way to report failure if the AudioSample is not rewindable;
* if the queued MIDI stream depletes during playback, the playback will stop, instead of waiting for new input. For real-time input, when the MIDI events arrive at the same speed that they're played, this would happen *after every event*!

Perhaps a new protocol, e.g. `AudioStream`, should be defined alongside `AudioSample`, and `Audio*Out` implementations will need to be updated to handle the new kind of input -- e.g. a new method `play_stream` which wouldn't take a `loop` parameter. Then `synthio` can have `MidiStream` alongside `MidiTrack`, with much of the implementation shared. Similarly, there can then be `RawStream` alongside `RawSample`.

chachagsedaro, 2023-04-07:
@tannewt has there been any development on this? i was curious to see if there's any way to implement small scale audio synthesis on circuitplayground bluefruit

dhalbert, 2023-04-07:
You might look at the recent PR's #7825 and #7840

closed 2023-05-22 by jepler