adafruit/circuitpython #4472: i2c not working properly unless I import wifi and set a small delay while initializing the i2c variable (See Video)

eddieespinal, 2021-03-24:
**Firmware**

```python
Adafruit CircuitPython 6.2.0-beta.1-dirty on 2021-03-22;
```

**Code/REPL**

```python
import time
import board
import busio
import wifi # without this import, i2c won't work
import adafruit_mpu6050

i2c = busio.I2C(board.SCL, board.SDA)
time.sleep(1) # without this delay, i2c won't work
mpu = adafruit_mpu6050.MPU6050(i2c, address=0x69)

while True:
    print("Acceleration: X:%.2f, Y: %.2f, Z: %.2f m/s^2" % (mpu.acceleration))
    print("Gyro X:%.2f, Y: %.2f, Z: %.2f degrees/s" % (mpu.gyro))
    print("Temperature: %.2f C" % mpu.temperature)
    print("")
    time.sleep(1)
```

**Behavior**

```python
code.py output:
Traceback (most recent call last):
  File "code.py", line 9, in <module>
  File "adafruit_mpu6050.py", line 144, in __init__
ValueError: No I2C device at address: 69
```

**Description**

- Error while using i2c with any i2c devices or sensors.
- Only happens when I don't include the import wifi and set a delay while initializing the i2c variable.

I tried using other i2c devices such as 128x64 display, same error same solution to make it work.

**Additional Info**

[Here is a video that shows what I'm describing here.](https://youtu.be/ACDmt20hbyc)

Adding the `import wifi` and `time.sleep(1)` solves the issue.

eddieespinal, 2021-03-24:
Thank you @dhalbert! - I will update it now and test this again.

eddieespinal, 2021-03-24:
Hi @dhalbert, I just updated to beta.4 and the issue is now fixed!.  Thank you again for your hard work and for replying so fast.  This ticket can be closed.

dhalbert, 2021-03-24:
Glad it works for you as well!

closed 2021-03-24 by dhalbert