adafruit/circuitpython #4476: Suggestion for adafruit_display_shapes.line and line thickness

bill22152, 2021-03-24:
Can we add a thickness to this function?  make it a default to 1 but allow user to set line width.

Also, can the line end in a Dot, Arrow, or nothing?

ladyada, 2021-03-24:
this is actually much harder than you'd think! line thickness needs to be handled as a low-level polygon drawing function.

kmatch98, 2021-03-24:
Currently one option is to use the vectorio library and draw a polygon (rectangle). I think it will created “filled” polygons only. 

If you need an example of a vectorio polygon look at the needle on the recent Dial widget here:  https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_Layout/blob/main/examples/displayio_layout_dial_simpletest.py

If you’d like other basic examples let me know and I can try to find one m

kmatch98, 2021-03-24:
Oh and here is the link to the vectorio documentation:
https://circuitpython.readthedocs.io/en/6.1.x/shared-bindings/vectorio/index.html

ladyada, 2021-03-24:
yeah a line that is not thickness 1 needs to be drawn as a filled poly!

jposada202020, 2021-03-29:
@I should have read this before, @kmatch98 and I were having a similar discussion here . https://github.com/adafruit/CircuitPython_Community_Bundle/pull/63
Maybe you could draw this in the python level!?

kmatch98, 2021-04-08:
I’d suggest the new library that @jposada202020 added to draw lines with arrows:

https://github.com/jposada202020/CircuitPython_ArrowLine

jposada202020, 2021-04-08:
I have added in the meantime a dot ending. It is available now in the Adafruit Community Bundle.

kmatch98, 2021-04-11:
This is addressed in the Python libraries and other code. Closing for now.

closed 2021-04-11 by kmatch98