adafruit/circuitpython #4477: btree for Circuitpython

bill22152, 2021-03-24:
Sometimes you just need indexed persistent storage without wifi.

ajs256, 2021-03-24:
You can write files to the `CIRCUITPY` drive from user code. [This guide](https://learn.adafruit.com/cpu-temperature-logging-with-circuit-python/) gives you the basics.

bill22152, 2021-03-25:
I need C.R.U.D. I get from btree or I could have used a text fileWm E Fitzpatrick(703) ***@***.***

dhalbert, 2021-03-25:
MicroPython has an optional `btree` module. Have you used it before? We have not turned it on in any CircuitPython builds.

bill22152, 2021-03-25:
I have used btree in 'Python', 'C' and 'C++' projects in the past and it works well at mapping hash arrays to storage..
It's light weight and easy to use.

Will MicroPython run on a PyPortal?

dhalbert, 2021-03-25:
MicroPython is not supported on the PyPortal. If you would like some kind of keyed storage immediately, you can use the `json` module and read and write JSON files, though it will not be as efficient as `btree`. We can leave this open if you would like something like `btree` in the long run.

closed 2021-03-25 by bill22152