adafruit/circuitpython #4479: [Pico] HID Mouse does not work on macOS Big Sur 11.2.1

bpatrzyk, 2021-03-24:
**Firmware**

```python
Adafruit CircuitPython 6.2.0-beta.4 on 2021-03-18; Raspberry Pi Pico with rp2040
```

**Code/REPL**

```python
import time
import board
import digitalio
import analogio
import usb_hid
from adafruit_hid.mouse import Mouse
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
mouse = Mouse(usb_hid.devices)
kbd = Keyboard(usb_hid.devices)
x_axis = analogio.AnalogIn(board.A0)
y_axis = analogio.AnalogIn(board.A1)
btn_left = digitalio.DigitalInOut(board.GP22)
btn_left.direction = digitalio.Direction.INPUT
btn_left.pull = digitalio.Pull.UP
pot_min = 0.00
pot_max = 3.29
step = (pot_max - pot_min) / 20.0
def get_voltage(pin):
    return (pin.value * 3.3)/65536
def steps(axis):
    return round((axis - pot_min) / step)
while True:
    x = get_voltage(x_axis)
    y = get_voltage(y_axis)
    print(x, y, btn_left.value)
    if steps(x) > 11.0:
        print('x>11', steps(x))
        mouse.move(x=10)
    if steps(x) < 9.0:
        mouse.move(x=-10)
    if steps(y) > 11.0:
        print('y>11', steps(x))
        mouse.move(y=10)
    if steps(y) < 9.0:
        mouse.move(y=-10)    
    if not btn_left.value:
        print('button')
        mouse.click(Mouse.LEFT_BUTTON)
        time.sleep(0.2)
    time.sleep(0.1)
```

**Behavior**

The mouse does not work on macOS Big Sur 11.2.1 but it works correctly on Windows 10 and Lubuntu 20.04.1.

**Description**
The board has an analog joystick (two potentiometers and a tact switch) connected. They are wired and configured correctly as in the console output I can see the correct voltages and switch status logged. What is more, the code above works correctly on Windows 10 and Lubuntu 20.04.1 - it moves the mouse cursor and performs the mouse clicks.
On macOS Big Sur 11.2.1 it has no effect at all - the cursor does not move and the click event does not work.

I tried adding the following keyboard key press code instead of the mouse click and it does work correctly on the Mac.
```
kbd.press(Keycode.A)  # "Press"...
kbd.release_all()  # ..."Release"!
```

Please let me know if there is anything I can do to help debugging or fixing this issue.

Removing [this](url) line resolves the issue.

dhalbert, 2021-03-24:
> Removing [this](url) line resolves the issue.

The url is not present, so I don't know which line you mean. Could you edit your post?

bpatrzyk, 2021-03-25:
> > Removing [this](url) line resolves the issue.
> 
> The url is not present, so I don't know which line you mean. Could you edit your post?

@dhalbert Thank you for the prompt response! I apologise, that comment was a leftover from the issue template that I missed and left there by accident. The rest of the issue description should be correct now.

dhalbert, 2021-03-31:
It appears that MacOS changed something that caused HID to stop working. This was already true with iOS: the presence of the gamepad HID device causes other HID devices to not work for iOS.

I will test this myself with a custom build. You can turn off the gamepad HID device if you want to try a custom build: https://learn.adafruit.com/building-circuitpython/customizing-usb-devices#customizing-usb-hid-devices-3084080-4

dhalbert, 2021-04-01:
> I apologise, that comment was a leftover from the issue template that I missed and left there by accident.

We took that out of the issue template; it was confusing, in this case both to you and me.

I just re-tested on High Sierra 10.13.6 and Big Sur 11.2.2. In neither case did the mouse work in the stock build, but if I set `CIRCUITPY_USB_HID_GAMEPAD = 0` in `mpconfigboard.mk` for the board I was using, the mouse **does** work. If you would like help with the custom build, feel free to ask here or in discord or the forums.

I thought in the past that HID Mouse did work on High Sierra, etc., but I must have misremembered. Perhaps it worked several years ago, when I was first working on the library. I know that with BLE HID, a Gamepad can be a problem on iOS.

dhalbert, 2021-04-01:
In CircuitPython 7.0.0 we plan to make it possible to turn these devices on and off in `boot.py`. In that case, we would probably also turn off `Gamepad` by default, in order to make `Keyboard` and `Mouse` work out of the box on all platforms.

bpatrzyk, 2021-04-03:
I am trying to make a custom build but I keep getting errors. I followed all setup instructions from the documentation and I tested it on macOS Big Sur and Ubuntu 20.04.

On Ubuntu I get the following error:
`ubuntu@circuitpython:~/circuitpython/ports/raspberrypi$ make BOARD=raspberry_pi_pico
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
QSTR updated
Traceback (most recent call last):
  File "gen_stage2.py", line 82, in <module>
    typer.run(main)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/typer/main.py", line 859, in run
    app()
  File "/home/ubuntu/.local/lib/python3.8/site-packages/typer/main.py", line 214, in __call__
    return get_command(self)(*args, **kwargs)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/typer/main.py", line 239, in get_command
    click_command = get_command_from_info(typer_instance.registered_commands[0])
  File "/home/ubuntu/.local/lib/python3.8/site-packages/typer/main.py", line 425, in get_command_from_info
    command = cls(  # type: ignore
TypeError: __init__() got an unexpected keyword argument 'no_args_is_help'
make: *** [Makefile:271: build-raspberry_pi_pico/stage2.c] Error 1`

On the mac I get the following error:
`➜  raspberrypi git:(main) ✗ make BOARD=raspberry_pi_pico
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
Traceback (most recent call last):
  File "gen_stage2.py", line 2, in <module>
    import cascadetoml
ModuleNotFoundError: No module named 'cascadetoml'
make: *** [build-raspberry_pi_pico/stage2.c] Error 1`

Please let me know if this is a known issue. I will keep experimenting with the build (perhaps build one of the release branches/tags instead of master).

What is more, I would suggest updating the MacOS Setup guide to replace the `brew install gcc-arm-embedded` with the manual installation (similar to the Linux setup). That is because the `gcc-arm-none-eabi-10-2020-q4-major-mac.tar.bz2` package is not signed and the MacOS prevents executing the compiler binary with the "Unidentified Developer" error. The `gcc-arm-none-eabi-10-2020-q4-major-mac.pkg` package is "Signed and notarized" and it works fine.

bpatrzyk, 2021-04-03:
I managed to make a build form the `6.2.0-beta.4` tag with the `CIRCUITPY_USB_HID_GAMEPAD = 0` flag and I confirm that the mouse does work on macOS Big Sur.

dhalbert, 2021-04-03:
> What is more, I would suggest updating the MacOS Setup guide to replace the `brew install gcc-arm-embedded` with the manual installation (similar to the Linux setup). That is because the [...]

Thanks for noting this! This is a moving target. I have updated the MacOS instructions in the Learn GUide.

daveythacher, 2021-04-04:
I am able to reproduce the Ubuntu issue. It does not work.

Neradoc, 2021-04-04:
> I just re-tested on High Sierra 10.13.6 and Big Sur 11.2.2. In neither case did the mouse work in the stock build

Tested on Catalina and Big Sur with a QT PY M0, with a code that jiggles the mouse for a sec on boot.

The mouse works in Circuitpython 6.2.0-beta2, but not any subsequent version available on S3. Intermediaries are missing due to cleanup, so maybe I'm gonna try to learn to use `git bisect`. Or lookup when tinyUSB was updated ?

dhalbert, 2021-04-04:
> The mouse works in Circuitpython 6.2.0-beta2, but not any subsequent version available on S3. Intermediaries are missing due to cleanup, so maybe I'm gonna try to learn to use `git bisect`. Or lookup when tinyUSB was updated ?

This was not what I expected when testing this; I thought it was only dependent on the HID descriptors, so I did not bother to test earlier verions. If you want to learn to bisect, go ahead, but I can track this down too.

dhalbert, 2021-04-04:
The descriptors differ. I'll work on this.

dhalbert, 2021-04-05:
Fixed by #4558 and #4560.

PR #4558 (2021-04-05): Put mouse before gamepad due to MacOS being finicky
Fixes #4479.

The HID devices were reordered after 6.2.0-beta.2 or so, and ended up being alphabetical. This put MOUSE last, which MacOS did not like. Reordering to KEYBOARD MOUSE ... GAMEPAD ... fixed the problem.

Tested on CPX because it was convenient:
```py
import time

import usb_hid
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
from adafruit_hid.mouse import Mouse

from adafruit_circuitplayground import cp

keyboard = Keyboard(usb_hid.devices)
mouse = Mouse(usb_hid.devices)

while True:
    if cp.b [... cut]
closed 2021-04-05 by dhalbert