adafruit/circuitpython #4482: [RP2040][I2C?] Feather RP2040 display garbage on 128x32 oled FeatherWing

dglaude, 2021-03-25:
Running Liz "I voted" badge on the Feather RP2040 and a FeatherWing 128x32 (I2C) I get garbage on the screen.
A video: https://photos.app.goo.gl/RrA6ntgNwnJnfuS16 

All libraries are up to date and run the latest firmware.
The same screen on any of my other Feather does work with the same code.
I believe I have had the same issue with a pico and manual wiring to the FeatherWing.

Other piece of code driving that screen on the Feather RP2040 did not display that kind of problem (but maybe I was getting strange CO2 reading from my SCD-30 (it was saying 2000+ but more in the 1000+ with another Feather).

I suspect some I2C issues... I would love to see someone with the same hardware reproduce that (to make sure it is not me or my hardware).

Found device at D:, running CircuitPython 6.2.0-beta.4-84-gc81007afb.
adafruit_debouncer==1.3.10
adafruit_displayio_ssd1306==1.3.1
adafruit_bitmap_font==1.3.7
adafruit_display_text==2.17.0
All is up to date.

dglaude, 2021-03-25:
Sorry, not from Liz, it is this guide: https://learn.adafruit.com/i-vote-d-pin

dhalbert, 2021-03-25:
Could you try `i2c = bitbangio.I2C(...)` and see if it works better? That would again point to an issue with how the RP2040 I2C hardware is being used.

dglaude, 2021-03-25:
It does trigger the error: "TimeoutError: Clock stretch too long"
```
# Set up display & add group
#i2c = board.I2C()
i2c= bitbangio.I2C(board.SCL, board.SDA) 
display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)
# ^^^ TimeoutError: Clock stretch too long
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=32)
group = displayio.Group(max_size=1)
display.show(group)

```

PS: Just in case I also checked inverting SDA/SCL but it give the same effect.

SAK917, 2021-03-25:
I recently filed a similar [issue with a Feather RP2040 displaying garbage on a 128x64 OLED FeatherWing](https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_SH1107/issues/4)

Please see the issue for details, and I found a workaround by using a bus speed of 400kHz instead of the default 100kHz using `i2c = busio.I2C(board.SCL, board.SDA, frequency=400000)`

It seems specific to the RP2040 as I tried it with a number of other boards and displays and only had the issue with the RP2040.

dhalbert, 2021-03-25:
@dglaude I am not seeing this problem with the following simpler test program. Could you try this one? I am testing with the absolute latest build (built locally): `6.2.0-beta.4-93-g0105b30a0`
```py
import board
import displayio
from adafruit_display_text import label
import adafruit_displayio_ssd1306
import terminalio
import time

displayio.release_displays()

i2c = board.I2C()
display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=32)
group = displayio.Group()
display.show(group)

# Add content to group
default_text = "Hello World\n"
counter = 0
text_area = label.Label(terminalio.FONT, text=default_text + str(counter), color=0xFFFFFF, x=0, y=8)
group.append(text_area)
display.show(group)

while True:
    time.sleep(1)
    counter += 1
    text_area.text = default_text + str(counter)
```

dglaude, 2021-03-26:
That "Hello World" count down does work fine with my
Adafruit CircuitPython 6.2.0-beta.4-84-gc81007afb on 2021-03-23; Adafruit Feather RP2040 with rp2040
(same as the one that fail on the "I Voted" that is problematic)

dhalbert, 2021-03-26:
@dglaude Could you try reproducing the "I Voted" bug, and then try this test firmware:
[feather-rp2040-i2c-sda-hold-time-fix.uf2.zip](https://github.com/adafruit/circuitpython/files/6214362/feather-rp2040-i2c-sda-hold-time-fix.uf2.zip)

dglaude, 2021-03-26:
Ok @dhalbert working on your request...
But first I wanted to confirm both my RP and the OLED are "OK".
This code was running before and still work fine: https://twitter.com/DavidGlaude/status/1375557719646740482?s=20
So there is an "issue" with the "I Voted" code (or something "different"):

dglaude, 2021-03-26:
@dhalbert That UF2 fix it for me... the I Voted code works great.
The hello world and my scd30 code also works.
Everything works!

dhalbert, 2021-03-26:
Great! That is the third device improved by this SDA hold time delay! Thanks for testing!

closed 2021-03-27 by ladyada

p5jokl, 2024-05-25:
> Ok @dhalbert working on your request... But first I wanted to confirm both my RP and the OLED are "OK". This code was running before and still work fine: https://twitter.com/DavidGlaude/status/1375557719646740482?s=20 So there is an "issue" with the "I Voted" code (or something "different"):

Hi David,
Interesting project! I'm trying to get the same display with Feather, FeatherWing and the SCD-30. Could you kindly share the code you used? I get it to work with a Metro board on a smaller OLED but not with the Feather/FeatherWing combination :(
Thanks so much.

[image]
[image]

PR #4499 (2021-03-27): Length RP2040 I2C SDA hold time
Incorporate https://github.com/adafruit/pico-sdk/pull/2, which lengthens SDA TX hold time.

This fixes I2C problems with the TCS34725 color sensor, and SSD1306 and SH1107 OLED driver chips.

Fixes #4482.
Fixes #4082.
Fixes #4498.