adafruit/circuitpython #4485: Is this a Bug or a Feature

bill22152, 2021-03-25:
```python
Adafruit CircuitPython 6.2.0-beta.2 on 2021-03-01; Raspberry Pi Pico with rp2040
```
mem_free = 137
Traceback (most recent call last):
  File "code.py", line 74, in <module>
  File "adafruit_display_shapes/rect.py", line 47, in __init__
MemoryError: memory allocation failed, allocating 19200 bytes

```python

     try:    

        Disp = displayio.Group(max_size=(20+3*(12+1)))
        # Blue Background
        Disp.append(Rect(0, 0, WIDTH, HEIGHT, fill=BLUE))  #line 47

...  lots of code ...

        gc.collect()
        freeMem = int(gc.mem_free()/1024)
        printf("\nmem_free = %d\n", freeMem)

        if freeMem > 32:
            board.DISPLAY.show(Disp)

    except Exception as e:
        eprintf("\n%s\n", str(e))
        pass

```
the Disp is in a while loop
I have tried to 'del Disp' before looping but no good

Two things
1)  why the memory allocation issue when so much ram is free?
2)  how come the pass didn't just continue running code.py

kmatch98, 2021-03-25:
Even if RAM is free, it may not have a large enough _contiguous_ "chunk" available for the size of the item you're trying to create.

If it will work for you, the best option is probably to use a `vectorio.Rectangle` since that won't need to allocate so much RAM for the `display_shapes.Rect` (that uses a bitmap).  If you are using a lot of shapes (polygons, rectangles, circles), consider using `vectorio` for all of them.  Here are the [docs for vectorio](https://circuitpython.readthedocs.io/en/6.1.x/shared-bindings/vectorio/index.html). 

Also, do you really have that many items (59) in your Group?  Seems like a lot.  If you're using a lot of text strings, it may help to use `bitmap_label.Label` instead of `label.Label`.

dhalbert, 2021-03-25:
This question is better asked in discord or the foruums. You are running out of RAM due to something, probably what's in `... lots of code ...`

dhalbert, 2021-03-25:
Also, as of 6.2.0-beta.3, a `Group` uses an unbounded list internally, so you can just remove the `max_size`; it is now ignored as of beta.3.

bill22152, 2021-03-25:
I could have 59 if the program would continue to run.... but its ending
before I get there..

Does the vectorio module have bitmap fonts?

kmatch98, 2021-03-25:
You can use bitmap_label for any text and then vectorio for polygons circles and rectangles. 

Sounds like you have a lot going on so you’ll have to search for any opportunities to save RAM.

bill22152, 2021-03-25:
Also, if I comment out the board.DISPLAY.show(Disp) line

The code runs as it should.  So all the Groups are being build and layered as they should,  It's only when I attempt to Show() that is dies, but it doesn't always die at the Show() most times it will loop and die when I am re-adding the groups.

kmatch98, 2021-03-25:
If you’re willing to share your code I have a base PyPortal here I can try it out on.

bill22152, 2021-03-25:
import gc
import time
import sys
import board
import busio
import adafruit_bme280
import displayio
from adafruit_display_shapes.rect import Rect
from adafruit_display_shapes.line import Line
from adafruit_display_text.label import Label
from adafruit_bitmap_font import bitmap_font

def printf(format, *args):
    sys.stdout.write(format % args)

def eprintf(format, *args):
    sys.stderr.write(format % args)

i2c = busio.I2C(board.SCL, board.SDA)
bme280 = adafruit_bme280.Adafruit_BME280_I2C(i2c)

# display = board.DISPLAY
WIDTH = board.DISPLAY.width
HEIGHT = board.DISPLAY.height

WHITE = 0xFFFFFF
GREY = 0x202020
BLUE = 0x191970
GREEN = 0x00ff00

#                 HEIGHT x WIDTH
# Titano     3.5″    320 x 480
# PyPortal   3.2"    240 x 320

font = bitmap_font.load_font("/fonts/Helvetica-Bold-36.bdf")
graphLeft = 141
graphHeight = 104

Samples = 12
graphXStep = int((WIDTH-graphLeft)/(Samples))

T = []
P = []
H = []

gc.collect()
printf("\nstarting ram = %d\n", int(gc.mem_free()/1024))

while True:

    try:

        # convert to F
        Tmp = int(round(10 * (bme280.temperature * 9 / 5 + 32), 0))
        # convert to InMg at 60°
        Prs = int(round(10 * bme280.pressure * 0.0296133971008484, 0))
        Hum = int(round(10 * bme280.relative_humidity, 0))

        if (len(T) > Samples):
            T.pop(0)
        T.append(Tmp)

        if (len(P) > Samples):
            P.pop(0)
        P.append(Prs)

        if (len(H) > Samples):
            H.pop(0)
        H.append(Hum)

        gc.collect()
        # Create Display bitmap
        Disp = displayio.Group(max_size=(20+3*(Samples+1)))

        # Blue Background
        Disp.append(Rect(0, 0, WIDTH, HEIGHT, fill=BLUE))

        # White grid
        Disp.append(Line(0, 1, WIDTH, 1, WHITE))
        Disp.append(Line(0, 106, WIDTH, 106, WHITE))
        Disp.append(Line(0, 212, WIDTH, 212, WHITE))
        Disp.append(Line(0, 319, WIDTH, 319, WHITE))
        Disp.append(Line(1, 0, 1, HEIGHT, WHITE))
        Disp.append(Line(140, 0, 140, HEIGHT, WHITE))
        Disp.append(Line(WIDTH-1, 0, WIDTH-1, HEIGHT, WHITE))

        # Black Graph area
        Disp.append(Rect(142, 2, 337, 104, fill=GREY))
        Disp.append(Rect(142, 107, 337, 104, fill=GREY))
        Disp.append(Rect(142, 213, 337, 104, fill=GREY))

        # Write Current in white.
        Color = WHITE
        Disp.append(Label(
            font, x=10, y=55, color=Color, text=("%.1f°" % (Tmp/10))))
        Disp.append(Label(
            font, x=10, y=160, color=Color, text=("%.1f%%" % (Hum/10))))
        Disp.append(Label(
            font, x=10, y=268, color=Color, text=("%.1f\"" % (Prs/10))))

        listStart = min(T)
        listRange = max(T) - listStart
        if listRange == 0:
            listRange = 0.0001
        listStep = ((graphHeight-3) / listRange)
        (x, y) = (graphLeft - graphXStep, 0)
        (x0, y0) = (0, 0)
        printf("Tmp=%d, len(T)=%d", Tmp, len(T))
        for i in T:
            printf(", %d", i)
            x += graphXStep
            y = graphHeight - 1 - int((i - listStart) * listStep)
            if (x0, y0) != (0, 0):
                Disp.append(Line(x0, y0, x, y, GREEN))
            (x0, y0) = (x, y)
        printf("\n")
        
        listStart = min(H)
        listRange = max(H) - listStart
        if listRange == 0:
            listRange = 0.0001
        listStep = ((graphHeight-3) / listRange)
        (x, y) = (graphLeft - graphXStep, 0)
        (x0, y0) = (0, 0)
        printf("Hum=%d, len(H)=%d", Hum, len(H))
        for i in H:
            printf(", %d", i)
            x += graphXStep
            y = graphHeight + 107 - int((i - listStart) * listStep)
            if (x0, y0) != (0, 0):
                Disp.append(Line(x0, y0, x, y, GREEN))
            (x0, y0) = (x, y)
        printf("\n")
        
        listStart = min(P)
        listRange = max(P) - listStart
        if listRange == 0:
            listRange = 0.0001
        listStep = ((graphHeight-3) / listRange)
        (x, y) = (graphLeft - graphXStep, 0)
        (x0, y0) = (0, 0)
        printf("Prs=%d, len(P)=%d", Prs, len(P))
        for i in P:
            printf(", %d", i)
            x += graphXStep
            y = graphHeight + 213 - int((i - listStart) * listStep)
            if (x0, y0) != (0, 0):
                Disp.append(Line(x0, y0, x, y, GREEN))
            (x0, y0) = (x, y)
        printf("\n")

        gc.collect()
        freeMem = int(gc.mem_free()/1024)
        printf("\nmem_free = %d\n", freeMem)

#        if freeMem > 32:
#            board.DISPLAY.show(Disp)

    except Exception as e:
        eprintf("\n%s\n", str(e))
        pass

    del Disp
    gc.collect()

    time.sleep(3*60)  # wait 3 minutes

bill22152, 2021-03-25:
`import gc
import time
import sys
import board
import busio
import adafruit_bme280
import displayio
from adafruit_display_shapes.rect import Rect
from adafruit_display_shapes.line import Line
from adafruit_display_text.label import Label
from adafruit_bitmap_font import bitmap_font

def printf(format, *args):
    sys.stdout.write(format % args)

def eprintf(format, *args):
    sys.stderr.write(format % args)

i2c = busio.I2C(board.SCL, board.SDA)
bme280 = adafruit_bme280.Adafruit_BME280_I2C(i2c)

# display = board.DISPLAY
WIDTH = board.DISPLAY.width
HEIGHT = board.DISPLAY.height

WHITE = 0xFFFFFF
GREY = 0x202020
BLUE = 0x191970
GREEN = 0x00ff00

#                 HEIGHT x WIDTH
# Titano     3.5″    320 x 480
# PyPortal   3.2"    240 x 320

font = bitmap_font.load_font("/fonts/Helvetica-Bold-36.bdf")
graphLeft = 141
graphHeight = 104

Samples = 12
graphXStep = int((WIDTH-graphLeft)/(Samples))

T = []
P = []
H = []

gc.collect()
printf("\nstarting ram = %d\n", int(gc.mem_free()/1024))

while True:

    try:

        # convert to F
        Tmp = int(round(10 * (bme280.temperature * 9 / 5 + 32), 0))
        # convert to InMg at 60°
        Prs = int(round(10 * bme280.pressure * 0.0296133971008484, 0))
        Hum = int(round(10 * bme280.relative_humidity, 0))

        if (len(T) > Samples):
            T.pop(0)
        T.append(Tmp)

        if (len(P) > Samples):
            P.pop(0)
        P.append(Prs)

        if (len(H) > Samples):
            H.pop(0)
        H.append(Hum)

        gc.collect()
        # Create Display bitmap
        Disp = displayio.Group(max_size=(20+3*(Samples+1)))

        # Blue Background
        Disp.append(Rect(0, 0, WIDTH, HEIGHT, fill=BLUE))

        # White grid
        Disp.append(Line(0, 1, WIDTH, 1, WHITE))
        Disp.append(Line(0, 106, WIDTH, 106, WHITE))
        Disp.append(Line(0, 212, WIDTH, 212, WHITE))
        Disp.append(Line(0, 319, WIDTH, 319, WHITE))
        Disp.append(Line(1, 0, 1, HEIGHT, WHITE))
        Disp.append(Line(140, 0, 140, HEIGHT, WHITE))
        Disp.append(Line(WIDTH-1, 0, WIDTH-1, HEIGHT, WHITE))

        # Black Graph area
        Disp.append(Rect(142, 2, 337, 104, fill=GREY))
        Disp.append(Rect(142, 107, 337, 104, fill=GREY))
        Disp.append(Rect(142, 213, 337, 104, fill=GREY))

        # Write Current in white.
        Color = WHITE
        Disp.append(Label(
            font, x=10, y=55, color=Color, text=("%.1f°" % (Tmp/10))))
        Disp.append(Label(
            font, x=10, y=160, color=Color, text=("%.1f%%" % (Hum/10))))
        Disp.append(Label(
            font, x=10, y=268, color=Color, text=("%.1f\"" % (Prs/10))))

        listStart = min(T)
        listRange = max(T) - listStart
        if listRange == 0:
            listRange = 0.0001
        listStep = ((graphHeight-3) / listRange)
        (x, y) = (graphLeft - graphXStep, 0)
        (x0, y0) = (0, 0)
        printf("Tmp=%d, len(T)=%d", Tmp, len(T))
        for i in T:
            printf(", %d", i)
            x += graphXStep
            y = graphHeight - 1 - int((i - listStart) * listStep)
            if (x0, y0) != (0, 0):
                Disp.append(Line(x0, y0, x, y, GREEN))
            (x0, y0) = (x, y)
        printf("\n")
        
        listStart = min(H)
        listRange = max(H) - listStart
        if listRange == 0:
            listRange = 0.0001
        listStep = ((graphHeight-3) / listRange)
        (x, y) = (graphLeft - graphXStep, 0)
        (x0, y0) = (0, 0)
        printf("Hum=%d, len(H)=%d", Hum, len(H))
        for i in H:
            printf(", %d", i)
            x += graphXStep
            y = graphHeight + 107 - int((i - listStart) * listStep)
            if (x0, y0) != (0, 0):
                Disp.append(Line(x0, y0, x, y, GREEN))
            (x0, y0) = (x, y)
        printf("\n")
        
        listStart = min(P)
        listRange = max(P) - listStart
        if listRange == 0:
            listRange = 0.0001
        listStep = ((graphHeight-3) / listRange)
        (x, y) = (graphLeft - graphXStep, 0)
        (x0, y0) = (0, 0)
        printf("Prs=%d, len(P)=%d", Prs, len(P))
        for i in P:
            printf(", %d", i)
            x += graphXStep
            y = graphHeight + 213 - int((i - listStart) * listStep)
            if (x0, y0) != (0, 0):
                Disp.append(Line(x0, y0, x, y, GREEN))
            (x0, y0) = (x, y)
        printf("\n")

        gc.collect()
        freeMem = int(gc.mem_free()/1024)
        printf("\nmem_free = %d\n", freeMem)

#        if freeMem > 32:
#            board.DISPLAY.show(Disp)

    except Exception as e:
        eprintf("\n%s\n", str(e))
        pass

    del Disp
    gc.collect()

    time.sleep(3*60)  # wait 3 minutes`

bill22152, 2021-03-25:
`
import gc
import time
import sys
import board
import busio
import adafruit_bme280
import displayio
from adafruit_display_shapes.rect import Rect
from adafruit_display_shapes.line import Line
from adafruit_display_text.label import Label
from adafruit_bitmap_font import bitmap_font

def printf(format, *args):
    sys.stdout.write(format % args)

def eprintf(format, *args):
    sys.stderr.write(format % args)

i2c = busio.I2C(board.SCL, board.SDA)
bme280 = adafruit_bme280.Adafruit_BME280_I2C(i2c)

# display = board.DISPLAY
WIDTH = board.DISPLAY.width
HEIGHT = board.DISPLAY.height

WHITE = 0xFFFFFF
GREY = 0x202020
BLUE = 0x191970
GREEN = 0x00ff00

#                 HEIGHT x WIDTH
# Titano     3.5″    320 x 480
# PyPortal   3.2"    240 x 320

font = bitmap_font.load_font("/fonts/Helvetica-Bold-36.bdf")
graphLeft = 141
graphHeight = 104

Samples = 12
graphXStep = int((WIDTH-graphLeft)/(Samples))

T = []
P = []
H = []

gc.collect()
printf("\nstarting ram = %d\n", int(gc.mem_free()/1024))

while True:

    try:

        # convert to F
        Tmp = int(round(10 * (bme280.temperature * 9 / 5 + 32), 0))
        # convert to InMg at 60°
        Prs = int(round(10 * bme280.pressure * 0.0296133971008484, 0))
        Hum = int(round(10 * bme280.relative_humidity, 0))

        if (len(T) > Samples):
            T.pop(0)
        T.append(Tmp)

        if (len(P) > Samples):
            P.pop(0)
        P.append(Prs)

        if (len(H) > Samples):
            H.pop(0)
        H.append(Hum)

        gc.collect()
        # Create Display bitmap
        Disp = displayio.Group(max_size=(20+3*(Samples+1)))

        # Blue Background
        Disp.append(Rect(0, 0, WIDTH, HEIGHT, fill=BLUE))

        # White grid
        Disp.append(Line(0, 1, WIDTH, 1, WHITE))
        Disp.append(Line(0, 106, WIDTH, 106, WHITE))
        Disp.append(Line(0, 212, WIDTH, 212, WHITE))
        Disp.append(Line(0, 319, WIDTH, 319, WHITE))
        Disp.append(Line(1, 0, 1, HEIGHT, WHITE))
        Disp.append(Line(140, 0, 140, HEIGHT, WHITE))
        Disp.append(Line(WIDTH-1, 0, WIDTH-1, HEIGHT, WHITE))

        # Black Graph area
        Disp.append(Rect(142, 2, 337, 104, fill=GREY))
        Disp.append(Rect(142, 107, 337, 104, fill=GREY))
        Disp.append(Rect(142, 213, 337, 104, fill=GREY))

        # Write Current in white.
        Color = WHITE
        Disp.append(Label(
            font, x=10, y=55, color=Color, text=("%.1f°" % (Tmp/10))))
        Disp.append(Label(
            font, x=10, y=160, color=Color, text=("%.1f%%" % (Hum/10))))
        Disp.append(Label(
            font, x=10, y=268, color=Color, text=("%.1f\"" % (Prs/10))))

        listStart = min(T)
        listRange = max(T) - listStart
        if listRange == 0:
            listRange = 0.0001
        listStep = ((graphHeight-3) / listRange)
        (x, y) = (graphLeft - graphXStep, 0)
        (x0, y0) = (0, 0)
        printf("Tmp=%d, len(T)=%d", Tmp, len(T))
        for i in T:
            printf(", %d", i)
            x += graphXStep
            y = graphHeight - 1 - int((i - listStart) * listStep)
            if (x0, y0) != (0, 0):
                Disp.append(Line(x0, y0, x, y, GREEN))
            (x0, y0) = (x, y)
        printf("\n")
        
        listStart = min(H)
        listRange = max(H) - listStart
        if listRange == 0:
            listRange = 0.0001
        listStep = ((graphHeight-3) / listRange)
        (x, y) = (graphLeft - graphXStep, 0)
        (x0, y0) = (0, 0)
        printf("Hum=%d, len(H)=%d", Hum, len(H))
        for i in H:
            printf(", %d", i)
            x += graphXStep
            y = graphHeight + 107 - int((i - listStart) * listStep)
            if (x0, y0) != (0, 0):
                Disp.append(Line(x0, y0, x, y, GREEN))
            (x0, y0) = (x, y)
        printf("\n")
        
        listStart = min(P)
        listRange = max(P) - listStart
        if listRange == 0:
            listRange = 0.0001
        listStep = ((graphHeight-3) / listRange)
        (x, y) = (graphLeft - graphXStep, 0)
        (x0, y0) = (0, 0)
        printf("Prs=%d, len(P)=%d", Prs, len(P))
        for i in P:
            printf(", %d", i)
            x += graphXStep
            y = graphHeight + 213 - int((i - listStart) * listStep)
            if (x0, y0) != (0, 0):
                Disp.append(Line(x0, y0, x, y, GREEN))
            (x0, y0) = (x, y)
        printf("\n")

        gc.collect()
        freeMem = int(gc.mem_free()/1024)
        printf("\nmem_free = %d\n", freeMem)

#        if freeMem > 32:
#            board.DISPLAY.show(Disp)

    except Exception as e:
        eprintf("\n%s\n", str(e))
        pass

    del Disp
    gc.collect()

    time.sleep(3*60)  # wait 3 minutes
`

kmatch98, 2021-03-25:
Ok, here's my first cut at it. Main changes:
- Separate the stagnant graphics (stored into the `background_Disp` group)and dynamic graphics (`foreground_Disp` group) into separate groups.
- I further broke `foreground_Disp` into three subgroups, each subgroup holds one set of measurement lines: `T_group`, `P_group` and `H_group`
- After each measurement, pop all the items out of the `T_group` and re-append all the lines.  Repeate with `P_group` and `H_group`.
- I replaced the stagnant rectangles and lines with `vectorio.Rectangle`

A few things you'll have to put back the way they were:
- I don't have the sensor so I just made it plot some random numbers for the data, you'll need to put back the sensor measurement code.
- I didn't have your font so I used another font I had lying around that is about the same size.

Without importing the sensor library, I seem to keep about 150kB of RAM free when looping for a while.  Let me know if something is unclear below.

```python

import gc
import time
import sys
import board
import busio
import adafruit_bme280
import displayio
from random import random
#from adafruit_display_shapes.rect import Rect
from adafruit_display_shapes.line import Line
from adafruit_display_text.bitmap_label import Label
from adafruit_bitmap_font import bitmap_font
from vectorio import VectorShape, Rectangle

#def printf(format, *args):
#sys.stdout.write(format % args)

#def eprintf(format, *args):
#sys.stderr.write(format % args)

i2c = busio.I2C(board.SCL, board.SDA)
#bme280 = adafruit_bme280.Adafruit_BME280_I2C(i2c)

WIDTH = board.DISPLAY.width
HEIGHT = board.DISPLAY.height

WHITE = 0xFFFFFF
GREY = 0x202020
BLUE = 0x191970
GREEN = 0x00ff00

font = bitmap_font.load_font("/fonts/DSEG14Classic-Regular-64-ModS.pcf")
graphLeft = 141
graphHeight = 104

Samples = 12
graphXStep = int((WIDTH-graphLeft)/(Samples))

T = []
P = []
H = []

gc.collect()
print("\nstarting ram = %d\n", int(gc.mem_free()/1024))

blue_palette = displayio.Palette(2)
blue_palette[1] = BLUE
grey_palette = displayio.Palette(2)
grey_palette[1] = GREY
white_palette = displayio.Palette(2)
white_palette[1] = WHITE

# background items do not change, so add them to a stagnant group called `background_Disp`

background_Disp = displayio.Group(max_size=20)
# background_Disp.append(Rect(0, 0, WIDTH, HEIGHT, fill=BLUE))
[... 139 lines cut ...]
            (x0, y0) = (x, y)
        print("\n")

        gc.collect()
        freeMem = int(gc.mem_free()/1024)
        print("\nmem_free = %d\n", freeMem)

        if freeMem > 32:
            board.DISPLAY.show(full_display_group)

        print('T_group size: {}'.format(len(T_group)))

    except Exception as e:
        print("\n%s\n", str(e))
        pass

    print("mem_free: {}".format(gc.mem_free()))

print('got here ***********')
#del Disp
gc.collect()

time.sleep(3*60)  # wait 3 minutes

```

kmatch98, 2021-03-25:
One more thing, the `display_shapes` library has a `Sparkline` in case you don't want to roll your own like you did here.

[Here are the Sparline docs.](https://circuitpython.readthedocs.io/projects/display-shapes/en/latest/api.html#adafruit_display_shapes.sparkline.Sparkline)

bill22152, 2021-03-25:
i played with my original code and placed a show() after each section
(background, Temp, Humid, Pres) and the issue went away...

Maybe to many groups at one time?  strange...

But now I'm switching to your code..  Thank you...

Also??  you didn't like my printf() function?

kmatch98, 2021-03-25:
Maybe doing `show()` cleared out some of the old memory references that you obliterated when redoing the Group, so maybe they can get freed up by the garbage collector.  Your guess is as good as mine.  Whatever works!

As for `printf`, that was new to me.  Rather than fight it, I traveled a known path.  Didn't want to get distracted from the core problem at hand.  But I will add something new to things I need to learn about.

closed 2021-03-26 by bill22152