adafruit/circuitpython #4491: Metro M4 Airlift board does not show up as CIRCUITPY drive after update

jackhwolf, 2021-03-25:
**Hardware**
Metro M4 Express Airlift [link](https://www.adafruit.com/product/4000?gclid=CjwKCAjw6fCCBhBNEiwAem5SOzdQZzPGQKoaNwQMdkN6rIpAJ2TJmaFa0Nl60JY81WoAUUVYuOrNQhoCifQQAvD_BwE)

**Firmware**
Contents of `INFO_UF2.TXT`: 
```
UF2 Bootloader v3.13.0 SFHWRO
Model: Metro M4 AirLift
Board-ID: SAMD51J19A-Metro-AirLift-v0
```
CircuitPython version `adafruit-circuitpython-metro_m4_airlift_lite-en_US-6.1.0.uf2`

**Issue**
Board does not show up as CIRCUITPY drive, even after going through multiple rounds of updating the bootloader and circuitpython version. Neopixel LED stays blue. This problem is happening on ~both~ of my Metro M4 Airlift boards but is not present on my CircuitPlayground Express or QTPY running the most current bootloader and circuitpython versions.

I have tried updating the bootloader and circuitpython distributions, using different cables, using different ports. I have been able to replicate this issue on multiple Metro M4 Airlift boards but it was not present with other boards. 

Any help would be much appreciated. Thank you in advance

dhalbert, 2021-03-26:
This is very unusual. I successfully loaded recent builds on a Metro M4 Express AirLift. Can you get to the REPL? Try to ctrl-C and try `import storage; storage.erase_filesystem()`, which will reformat CIRCUITPY.

What kind of host computer are you using, and what version of the operating system?

If the reformat doesn't help, I'd suggest two more things::
1. If it's Windows, clear out the USB devices using the USB Device Cleanup tool from here: https://www.uwe-sieber.de/misc_tools_e.html. Don't have the board plugged in when your are doing it
2. Try a different computer. If you have a Raspberry Pi, that would be an easy thing to try.

bludin, 2021-03-26:
I just faced the same problem with my PyPortal Titano under Win10. Looking in the Device Manager, I found a USB Mass Storage device that was disabled. I enabled it and CIRCUITPY mounted instantly.

dhalbert, 2021-03-26:
@bludin Thanks for the tip! Do you have any idea how that happened? Was there some USB error at some point and Windows may have disabled it?

bludin, 2021-03-26:
@dhalbert I  didn't notice anything along these lines. The only reason why I even looked in that place is because the description of the  USB Device Cleanup tool mentioned that you could also delete USB devices one by one from the Device Manager.

jackhwolf, 2021-03-26:
I am running Ubuntu 20.04. Now that I think about it this is the first update I’ve done w this OS on these boards, but like I said worked for some boards

dhalbert, 2021-03-26:
@jackhwolf I also tested on Ubuntu 20.04 Have you tried the `storage.erase_filesystem()` yet, and have you rebooted the host machine between trials?

jackhwolf, 2021-03-26:
Is there a way for me to access storage.erase_filesystem() without being able to access the device repl?

ladyada, 2021-03-26:
https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#old-way-for-the-circuit-playground-express-feather-m0-express-and-metro-m0-express-2978458-27
use metro m4

jackhwolf, 2021-03-28:
@ladyada , thank you so much! Everything is working now. Is there any way for me to figure out where I went wrong/what happened?

closed 2021-03-28 by ladyada