adafruit/circuitpython #4492: Reboot loop when using Adafruit RFM9x library

AdamLeyshon, 2021-03-26:
**Firmware**
```
adafruit-circuitpython-adafruit_feather_rp2040-en_US-6.2.0-beta.4.uf2
adafruit-circuitpython-bundle-6.x-mpy-20210324
```

```python
Adafruit CircuitPython 6.2.0-beta.4 on 2021-03-18; FeatherS2 with ESP32S2
```

**Code/REPL**

```python
import board
import busio
import adafruit_rfm9x

spi = busio.SPI(clock=board.IO36, MOSI=pin.IO37, MISO=board.IO35) # Init SPI Bus
print("SPI bus INIT OK")

rfm9x = adafruit_rfm9x.RFM9x(spi=spi, cs=board.IO5, reset=board.IO6, frequency=915.0)
```

**Behavior**
`SPI bus INIT OK` is printed followed by the board crashing and entering a reboot loop when `adafruit_rfm9x.RFM9x()` is called.

I highly suggest adding a sleep at the start to allow you time to connect to the REPL.

Using the predefined `board` SDI/SDO/SCK pins makes no difference either or even `board.SPI()`
No matter which pins I pick for CS and Reset, the crash is the same.

AdamLeyshon, 2021-03-26:
Please disregard.

The FeatherS2 has labels SDI and SDO on the board, but when connecting the RFM module, they must connect to the opposite:
`SDI -> MISO` and `SDO -> MOSI`

Where as on the Pico and some other boards it's `MOSI -> MOSI` and `MISO -> MISO`.

Which sort of makes sense, `In -> Out`, `Out -> In`

In my case, I have simply flipped the pins in code:
```python
import board
import busio
import digitalio
import adafruit_rfm9x
import feathers2

# Make sure the 2nd LDO is turned on
feathers2.enable_LDO2(True)

time.sleep(2) # Allow time to intertupt boot

cs = digitalio.DigitalInOut(board.IO5)
reset = digitalio.DigitalInOut(board.IO6)

spi = busio.SPI(board.SCK, MOSI=board.MISO, MISO=board.MOSI) # Init SPI Bus
print("SPI INIT OK")

rfm9x = adafruit_rfm9x.RFM9x(spi, cs, reset, frequency=915.0)
rfm9x.auto_agc = True
rfm9x.output_power = 23

print("RFM INIT OK")
```

Gives 
```
SPI INIT OK
RFM INIT OK
```

closed 2021-03-26 by AdamLeyshon

ladyada, 2021-03-26:
@unexpectedmaker please add to your documentation!

jerryneedell, 2021-03-26:
I'm not sure I see the problem -- SDI is Serial Data In  so in the Master device this is MISO - Master In Slave Out