adafruit/circuitpython #4494: another problem with busdevice-in-core and bitbangio?

jepler, 2021-03-26:
I found that with the following code, the with-statement is never entered (circuitpython waits forever).  However, if I change the type of spi bus to board.SPI() aka busio.SPI, it completes.

```python
# spi = board.SPI()
spi = bitbangio.SPI(board.SCK, MISO=board.MISO, MOSI=board.MOSI)
cs = digitalio.DigitalInOut(board.D10)
dev = adafruit_bus_device.spi_device.SPIDevice(spi, cs)
print("before with-statement")
with dev as card:
    print("with-statement")
print("after with-statement")
```

jepler, 2021-03-26:
@ladyada this is one of the problems that prevents bitbangio to work with adafruit_sdcard, the other is that the constant write value during an SPI read transaction is not supported.

ladyada, 2021-03-26:
ok!

PR #4497 (2021-03-26): Spi enhancements
This required both adding support for the "default write value" in bitbangio.SPI, and also supporting duck-typing in adafruit_bus_device's SPIDevice which, surprisingly, was overlooked when we fixed I2C.

Now, the following code works, tested on RT1010-EVK:

```python
import os
import board
import bitbangio
import adafruit_sdcard
import storage
import digitalio

spi = bitbangio.SPI(board.SCK, MISO=board.MISO, MOSI=board.MOSI)
cs = board.D10
sdcard = adafruit_sdcard.SDCard(spi, digitalio.DigitalInOut(cs))
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
print(os.listdir(" [... cut]
closed 2021-03-26 by ladyada