adafruit/circuitpython #4502: Hardfault when picking wrong UART pin on iMX RT1011

ladyada, 2021-03-27:
```
import board
import busio

uart = busio.UART(None, board.TX, baudrate=9600)

while True:
   pass
```
i should be passing board.RX in there, was testing if the wrong pin is passed...it hardfaults when instead we should get an exception that the pin cannot be used as a uart RX!

alecharp, 2022-01-31:
@ladyada sorry to ping you here but I'm trying to understand this in the context of using a KB2040 on a split keyboard. Both half should be communicating in UART and are linked on the Rx pin. With https://github.com/KMKfw/kmk_firmware/blob/e70ce5f4315db02beb25302f4ce027f739fba99b/kmk/modules/split.py#L111-L118, it means that the (default) left half uses the Rx pin as a receiver end, and the right half uses the Rx pin as the transmitter pin. 

Is that code (KMK split module) broken with #4513 fix or is there something else I completely misunderstood? 

I'm asking because, even without KMK, as with the code 

```
import board, busio
busio.UART(tx=board.RX, rx=None)
```
I get the `invalid pin` error. 

Thank you for your time.

ladyada, 2022-01-31:
@alecharp hi please do not post on an unrelated, closed issue
you can create a new issue if there is a circuitpython bug :)

alecharp, 2022-01-31:
Sorry, didn't see it was unrelated. Creating a new ticket.

PR #4513 (2021-03-30): mimxrt1011: UART: Add additional error checking
.. and make the 'invalid pin' messages standard.

Closes #4502
commit 71eee45a (2021-03-30): mimxrt1011: UART: Add additional error checking

.. and make the 'invalid pin' messages standard.

Closes #4502