adafruit/circuitpython #4506: Pulseio not working properly on RAspberry Pi Pico M2040

luceec, 2021-03-29:
Test Case:
Want to measure duty cycle and convert to RGB values

Test Setup:
PWM input from RGB LED conroller connected to pin 18 on Pico (used a level converter to bring 5V down to 3.3V)
Using this version on the Pico "Adafruit CircuitPython 6.2.0-beta.4-116-g010232574 on 2021-03-27; Raspberry Pi Pico with rp2040"
Square wave is a 7.83 KHz

Test program:
`import board, import pulseio`
`pulses = pulseio.PulseIn(board.GP18, maxlen=2, idle_state=True)`
`print(len(pulses))`

Error:
REPL becomes unresponsive.
>>> ERROR   thonny.plugins.micropython.backend: PROBLEM WITH THONNY'S BACK-END:
Traceback (most recent call last):
  File "C:\Users\lucee\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 340, in _handle_normal_command
    response = handler(cmd)
  File "C:\Users\lucee\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 636, in _cmd_get_globals
    "{name : (__thonny_helper.repr(value), id(value)) for (name, value) in globals().items() if not name.startswith('__')}"
  File "C:\Users\lucee\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 549, in _evaluate
    out, err = self._execute(script, capture_output=True)
  File "C:\Users\lucee\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 505, in _execute
    self._execute_with_consumer(script, consume_output)
  File "C:\Users\lucee\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 708, in _execute_with_consumer
    self._submit_code(script)
  File "C:\Users\lucee\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 572, in _submit_code
    self._submit_code_via_raw_mode(to_be_sent)
  File "C:\Users\lucee\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 622, in _submit_code_via_raw_mode
    self._write(block)
  File "C:\Users\lucee\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 517, in _write
    self._connection.write(data)
  File "C:\Users\lucee\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\serial_connection.py", line 85, in write
    assert size == len(data)
AssertionError

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

DavePutz, 2021-03-29:
I was not able to reproduce the hang using PWM out at 7.83 KHz from one Circuitpython board into a Pico.
A few comments:
1) The stack output doesn't show us anything - it is a thonny stack and not from circuitpython
2) The print of len(pulses) probably doesn't show what you think it does, pulsein needs calls to popleft() to get values. So, you
     may want to do something like:
```
...
pulses_in=[]
while pulses:
    pulses_in.append(pulses.popleft())
print(len(pulses_in), "pulses_in:", [x for x in pulses_in])
```
3) The maxlen parameter determines how many pulses you wish to capture. Do you really only want 2?
4) Note that pulsein.popleft() will return both high and low pulses. So if you only want to capture the length of 'high' pulses you will want to skip every other value.
5) You may want to check one of the libraries that use pulsein (such as IR remote at https://github.com/adafruit/Adafruit_CircuitPython_IRRemote/blob/master/adafruit_irremote.py) for an example

tannewt, 2021-03-29:
I believe there is a CircuitPython backend to Thonny that will work better than the MicroPython one. The MicroPython one "runs" the code through the REPL which we don't really support. Instead, we expect the code to execute to be written to the code.py file on the disk. Mu is an alternative that does this: https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython

closed 2021-03-29 by tannewt