adafruit/circuitpython #4510: supervisor.runtime.run_reason has the wrong code after supervisor.reloadI()

mikejc58, 2021-03-30:
**Firmware**

This occurred on 
Adafruit CircuitPython 6.1.0 on 2021-01-21; Adafruit Matrix Portal M4 with samd51j19

**Code/REPL**

Here is a sample program that demonstrates the problem and my boot.py

**code.py**

```
import supervisor
import time

run_reason = supervisor.runtime.run_reason 
time.sleep(1)
print("run reason == {}".format(run_reason))
print("Exit via supervisor.reload()")
supervisor.reload()
```
**boot.py**

```
import supervisor
import storage

supervisor.disable_autoreload()
storage.remount("/", False)

```

**Behavior**
```
run reason == supervisor.RunReason.STARTUP
Exit via supervisor.reload()

Code stopped by auto-reload.
soft reboot

Auto-reload is off.

code.py output:
run reason == supervisor.RunReason.AUTO_RELOAD
Exit via supervisor.reload()

Code stopped by auto-reload.
soft reboot

Auto-reload is off.

code.py output:
run reason == supervisor.RunReason.AUTO_RELOAD
  .
  .
  .

```

**Description**

After a supervisor.reload(), supervisor.runtime.run_reason has the code supervisor.RunReason.AUTO_RELOAD,  rather than the supervisor.RunReason.SUPERVISOR_RELOAD code expected.  Also note that the text that shows up on the USB console 'Code stopped by auto-reload' suggests that CircuitPython thought there was a new code.py stored...but there was not, and auto-reload is off.

**Additional Info**

Removing [this](url) line resolves the issue.

anecdata, 2021-03-30:
Related: #3858

tannewt, 2021-03-30:
@hierophect want to take a look? You've been in this code most recently.

PR #4708 (2021-05-05): Remove overwrite of run_reason in main
This PR fixes an issue in main that would cause any pre-existing run reason, such as SUPERVISOR_RELOAD set in `supervisor.reload()`, to be overwritten after the `reload_requested` trigger. `autoreload.c` now sets the run reason as AUTO_RELOAD itself.

Resolves #4510
Resolves #3858
closed 2021-05-05 by hierophect