adafruit/circuitpython #4531: Disable HID Gamepad by default once dynamic HID descriptors are implemented

dhalbert, 2021-04-01:
HID Gamepad causes MacOS and iOS not to respond to HID Mouse. in 7.0.0, disable HID Gamepad by default, assuming that dynamic HID descriptors are implemented. Addresses #4479.

hathach, 2021-04-08:
I am late to the party, did re-arrange #4560 the report ID get both mouse and gamepad work with iOS. I have seen similar issue with BLE HID as well on the arduino nrf52. Plan to get back to implement it in the near future. Another way it adding additional HID interface for gamepad, it may work.

dhalbert, 2021-04-23:
@hathach #4560 fixed it on MacOS but not iOS. I am deep into dynamic USB descriptors now, so it should be possible. I will not implement two HID devices yet, but maybe eventually.

dhalbert, 2021-05-21:
Fixed by #4689.

closed 2021-05-21 by dhalbert