adafruit/circuitpython #4532: Mouse input does not work (is ignored) in MacOS

marcos-diaz, 2021-04-01:
## Setup
* Firmware: CircuitPython 6.2.0 beta 3 (en_US)
* Board: Raspberry Pi Pico
* Host: Apple MacOS Big Sur 11.2.2 on MacBook Pro 2014

## Reproduce
1. Load this code as `code.py`
```
import time
import board
from digitalio import DigitalInOut, Pull
import usb_hid
from adafruit_hid.mouse import Mouse

button = DigitalInOut(board.GP16)
button.pull = Pull.DOWN
mouse = Mouse(usb_hid.devices)

while True:
    time.sleep(0.1)
    if button.value:
        print('CLICK!')
        mouse.click(Mouse.LEFT_BUTTON)
```

2. Connect to host OS via USB cable.
3. Push the button.

## Expected results
* Mouse clicks are performed on pushing the button.
* It should work in any operative system.

## Actual results
* MacOS recognises there is a mouse (see below), but fails to perform a mouse click.
* It does not work either in MacOS recovery mode (no user config).
* Works as expected on Windows 10.
* Works as expected on GNU/Linux.
* Works as expected on PC BIOS.
* No errors nor warnings are shown in the stdout.

## Additional notes.
* Mouse movement and wheel does not work either, with same results regarding OS' (it works in everything but MacOS).
* Mouse `usb_hid.device` is present as expected with 0x01 page and 0x02 usage.
* MacOS actually detects there is a mouse connected since the mouse config menu goes from this
[image]
to this
[image]
when connected.

Thanks.

dhalbert, 2021-04-01:
I got this issue mixed up with #4479. The report of beta.2 working is surprising, and I will see what is going on. Nevertheless, see #4531.

dhalbert, 2021-04-01:
Aha: the YouTube video https://www.youtube.com/watch?v=MjCFJCfq8ko cited above shows a mouse being used on Linux. This is a MacOS-specific problem. Closing this in favor of #4479, but they were both helpful.

closed 2021-04-01 by dhalbert

dhalbert, 2021-04-05:
Fixed by #4558 and #4560.