adafruit/circuitpython #4540: Interrupts as Exceptions

dhalbert, 2021-04-02:
Here's an idea about using Python exceptions to allow something like a button push or other external interrupt to interrupt normal program flow, without having to provide general interrupt handlers or having to poll.

For example, suppose you want a button push to interrupt a NeoPixel animation, and go on to the next animation. Instead of checking for the button push at various strategic places, we could just ask that a transition `DigitalInOut` cause an exception to be raised. The outer loop of the program would catch this exception and move on to the next animation. Here is strawman code:

```py
import board
import digitalio

button = digitalio.DigitalInOut(board.D0)
button.pull = digitalio.Pull.UP
# Might need debouncing support here.
button.raise_exception_on_transition(digitalio.Transition.DOWN)

NUM_ANIMATIONS  = 5
animation_number = 0

while True:
    try:
        while True:
            # Animation can be interrupted at any time,
            # not just when it finishes.
            run_animation_once(animation_number)
    except digitalio.TransitionException
        animation_number = (animation_number + 1) % NUM_ANIMATIONS
```
This is clean and simple. The animation code needs to be able to deal with being interrupted at any time, but it probably has no interesting internal state that would be messed up by an exception.

Additional features could be that you can specify the exception, that it passes the `DigitalInOut` that caused it back so you can handle multiple buttons, that it does debouncing, etc.

Unlike an interrupt handler, the interrupt happens "in-line", because it's an exception. So you can't continue from where you left off, but often you don't want to anyway.

dhalbert, 2021-04-02:
The same idea could also be used with timers. A timer would go off and raise an exception. I'm not sure where to put that functionality. Both the pin capability above and the timer capability could also be part of `alarm`, but for pins, the above seems cleaner. Or perhaps `alarm` might be refactored in some way.

ladyada, 2021-04-02:
im intrigued - mostly because i think it would be an easy-ish (comparatively) thing to do!

jepler, 2021-04-02:
Here are some initial thoughts.

 * It risks running afoul of places where people wrote 'except Exception as e:', unless this new kind of exception is outside of the Exception class hierarchy (like KeyboardInterrupt is).
 * More generally, there is a lot of code which assumes that exceptions can only happen at known moments, except for program-terminating exceptions like KeyboardInterrupt generally is. We could fix the code once we know about it, of course.
 * Would the 1ms tick need to be running anytime a `raise_exception_on_transition` object is around?
 * Can multiple buttons be used in this way at the same time? does the exception tell you which one (for instance by returning the DigitalInOut instance as a property of the exception)?
 * Is there a way to ensure there's not a window of opportunity for the button to be pressed between the `raise_exception_on_transition()` call and the `try` block?

For the last point
```python
try:
    with button.raise_exception_on_transition(digitalio.Transition.DOWN):
        ...
except digitalio.TransitionException as e:
    ...
```
ensures the duration of the `raise_exception_on_transition` object is less than the `try/except` block.  `as e` implies that the exception could have useful content.

dhalbert, 2021-04-02:
> * It risks running afoul of places where people wrote 'except Exception as e:', unless this new kind of exception is outside of the Exception class hierarchy (like KeyboardInterrupt is).

I think `except Exception as e:` is kind of poor practice, and presumably someone using ` TransitionException` would know that would happen.

> * More generally, there is a lot of code which assumes that exceptions can only happen at known moments, except for program-terminating exceptions like KeyboardInterrupt generally is. We could fix the code once we know about it, of course.

Yes, it would have to be kept in mind. We already have to deal with lots of potential exceptions in network code, so we are somewhat used to this. This is a lot more controlled.

Some people say that using exceptions for program flow control is an anti-pattern. I agree if there are simpler ways (like passing back error codes), that it's better not to use an exception. But Python uses exceptions quite liberally for situations that are not necessarily unexpected errors (e.g. use of `KeyError`), so I don't feel too bad about that. And this is too simple to pass up, I think.

> * Would the 1ms tick need to be running anytime a `raise_exception_on_transition` object is around?

No, it would be a real interrupt internally, not polling, so the `WFI()` would be triggered.

> * Can multiple buttons be used in this way at the same time? does the exception tell you which one (for instance by returning the DigitalInOut instance as a property of the exception)?

>> [from the first post:] Additional features could be that you can specify the exception, that it passes the DigitalInOut that caused it back so you can handle multiple buttons, that it does debouncing, etc.

dhalbert, 2021-04-02:
If all one cares about is the first transition, debouncing can be handled in the `except:`
```py
    except digitalio.TransitionException
        animation_number = (animation_number + 1) % NUM_ANIMATIONS
        time.sleep(0.1)  # debounce
```

jepler, 2021-04-02:
> If all one cares about is the first transition, debouncing can be handled in the `except:`
> 
> ```python
>     except digitalio.TransitionException
>         animation_number = (animation_number + 1) % NUM_ANIMATIONS
>         time.sleep(0.1)  # debounce
> ```

Is a TransitionException a one-time event?  If not, it can re-trigger inside the Except.

ladyada, 2021-04-02:
(btw, i dont really trust transition irq s - they can get lost. i like level type :)

dhalbert, 2021-04-02:
> Is a TransitionException a one-time event? If not, it can re-trigger inside the Except.

Good point. Details to be worked out.

dhalbert, 2021-04-02:
Safer if one time:
```py
while True:
    # Triggers once only. Re-enabled each time around the loop.
    button.raise_exception_on_transition(digitalio.Transition.DOWN)
    try:
        while True:
            # Animation can be interrupted at any time,
            # not just when it finishes.
            run_animation_once(animation_number)
    except digitalio.TransitionException
        animation_number = (animation_number + 1) % NUM_ANIMATIONS

microDev1, 2021-04-02:
+1 for this. Why not point the existing `alarm` api to raise an exception. We already use interrupt in fake deep sleep.
```python
import alarm
import board

pin_alarm = alarm.pin.PinAlarm(board.IO0, edge=True)

while True:
    # Triggers once only. Re-enabled each time around the loop.
    alarm.raise_exception_on_alarms(pin_alarm)
    try:
        while True:
            # loop
    except alarm.PinAlarmException
        # handle interrupt

```

microDev1, 2021-04-03:
I did some more work on this and came up with the following api:

```python
import alarm
import board

touch_alarm = alarm.touch.TouchAlarm(board.IO6)
alarm.raise_exception_on_alarms(touch_alarm)

try:        
    while True:
        pass
        # loop
except alarm.AlarmException as exc:
    if exc.args[0] == touch_alarm:
        # handle interrupt
        print(exc.args[0].pin)
```

This is similar to the existing `alarm` api and also reduces the exception to only one (i.e. `AlarmException`).
I have pushed the initial implementation to my [`interrupt-exception`](https://github.com/adafruit/circuitpython/compare/main...microDev1:interrupt-exception) branch.

dhalbert, 2021-04-03:
@microDev1 Thanks for doing a proof of concept so fast! I am interested in having these even in small builds, so we have to think about how to make `alarm` pervasive and still fit. It may not be possible. Perhaps we would leave the sleep functionality out, but I think we may still have to remove something else from small builds to make it fit.

WarriorOfWire, 2021-04-04:
Oh man I’m pretty sure this is a bad plan and there is so much to respond to 😅 

How do circuitpython library developers make reliable libraries if random unrelated exceptions may be thrown at any point in circuitpython code? Are you going to ask every python library that exists to disable interrupt exceptions “just in case” for critical sections of code?

Okay, forget library developers for a second, how on earth do you teach users what is safe to put inside of a `with button.raise_on_press()`? There are simpler choices and a justification that “circuitpython uses exceptions so we can use them here too” does _not_ demonstrate that they _should_ be used here.

__Interrupts should be handled in normal program flow.__ They are not exceptional in circuitpython or really in any language! In other languages you get a separate tiny stack for exceptions - throwing away your single circuitpython stack _is not the same thing_ and that’s what exception style interrupts do.

Toy while true: pass loops make it look reasonable but if you are doing anything interesting the exception approach is unusable. Try initializing a tft where you hit the reset pin, sleep for 60ms, unflip the pin, sleep for 160ms, send configuration commands, sleep for 20ms, set the backlight. Do you want to write a state machine that covers every line of that so if a button is pushed during that time your screen initialization can be resumed? You need a state for every line. More than that even, you need to log an intent per line _and_ you need to somehow be able to discover whether you effected that line previously because you can’t transactionally write a variable and send data over I2C.

What about displayio? If you’re trying to put 6 things on the display and you just get unrecoverably yanked out of your control flow through no fault of your code, it’s a bug in circuitpython. I.e., this feature is a bug.

Instead, let developers choose when to handle/dispatch interrupts.
```
// array of interrupt counters, reset at get time.
interrupts = interrupts.get_interrupts()
for interrupt in (i for i in interrupts if i.count > 0):
  handle_interrupt(interrupt)
```

Or you could have the api at interrupt configuration time take a handler function, and that above loop could be in C and your while True would have a call to `handle_interrupts()` where the developer wants it. Fun fact, this style would help enable a highly efficient `asyncio` `park_for_interrupts()` as well as the obvious, plain and easy while true polling approaches for users.

With the original proposal here, any program of any complexity will need something like a no-interrupts context manager and the main loop will look like
```
while True:  # an exercise for the reader, how many wrapped `while True: try: while True: try while True`s would you need to ensure your program didn’t crash because your root while true is outside of a try and an interrupt is possible while the interpreter is evaluating your `while true`? Alternatively, how do you account for the possibility of an interrupt raising an exception on you while you are handling the previous interrupt in your except?
  try:  # so when the context manager falls off we don’t end the program
    with NoInterrupts():
      do_business_logic()
      update_display()
      poll_devices()
      update_lighting()
    # hilariously, handled interrupts fire on this comment line
  except Interrupt as e:
    handle_interrupt(e)  # if another interrupt raises in here the application terminates
```

or you’d need to have a loop that does the 4 actual loop lines above, then dives into a try: enable_interrupts(); disable_interrupts(); catch Interrupt as e: disable_interrupts(); handle_interrupt(e). Yeah this would work and would approximate an incredibly complicated version of the simple non-exception style of pollable interrupt counters.  Oh wait, actually this doesn’t work because between catching the exception and disabling interrupts you can have another interrupt arrive and raise on you. If you auto-disable interrupts after firing 1 until they’re re-enabled, are you maintaining counters or is the expectation that this would be useless for pulse counting? The exception design is so subtle I submit that it’s a footgun at the very best and more likely just broken.

Please just let developers ask how many times a particular interrupt has fired since the last time they asked about it. Interestingly, it’s a much easier user-facing api to design; importantly, it is consistent with circuitpython and simple stacks; critically, it’s __incomparably easier__ for end users to use correctly.

dhalbert, 2021-04-04:
@WarriorOfWire I think you make a good case it is in general a bad idea, and too dangerous. :smile: This was an early-morning thought bubble that does not survive sunlight.

I was thinking of a rather specific case where the operation being terminated abruptly is safe to do so, because it is idempotent or similar. NeoPixel animations are a perfect example. They keep some internal state that can be discarded, and can be stopped at any time (at a Python call boundary, which is what this does.) I did not mean to suggest that this is a general solution as to how to handle interrupts. But as you point out, it requires the end user to know when this technique would be safe.

I will close this and continue the quest for simple idioms of handling asynchronous events that can be explained to the beginning programmer. You have suggested several interesting ideas above.

closed 2021-04-04 by dhalbert

deshipu, 2021-04-05:
What happens when your button exception gets rised again inside your exception-handling code? Or is this a one-shot thing?