adafruit/circuitpython #4541: Finish PWM on mimxrt1011

jepler, 2021-04-02:
Basic PWM has been enabled, but there are still limitations.  Here are the ones I know of:
- [ ] never-reset PWMs (like for backlight)
- [ ] correct checking for PWM sharing
- [ ] support for "X-channel" PWM, if possible (no code in SDK for it) - this affects only pin A4 (and ESP_READY) on the metro m7 1011

tannewt, 2023-02-15:
I don't think I'll add X channel support because it'll limit frequencies to the prescaler capable ones. VAL1 is used for top value usually but then it leads to a low pulse every cycle on X. This would prevent 100% duty cycle.

[image]

tannewt, 2023-02-22:
Nevermind, I added X support but it'll raise an error on 100% duty cycle. Note, it also has a fixed falling edge when most PWM are fixed leading edge. The CircuitPython API doesn't explicitly say which it is.

commit 1acf65ee (2023-02-28): Fix `pwmio` on iMX RT.

It now handles deinit, never_reset and sharing tracking. PWM
now runs in the WAIT state as well during a time.sleep().

_reset_ok() was removed because it was called in one spot right
before deinit().

Some PWMOut were also switched to a bitmap for use instead of
reference count. That way init and deinit are idempotent.

Fixes #6589. Fixes #4841. Fixes #4541.
closed 2023-02-28 by dhalbert