adafruit/circuitpython #4542: Interrupts: what are your use cases?

dhalbert, 2021-04-02:
People ask if CircuitPython supports interrupts. It does not support user-written interrupt handlers in the way that MicroPython or Arduino do. However, it of course does use interrupts internally for specific functionality, such as `PulseIn`, `countio`, `rotaryio`, etc., etc.

MicroPython supports [`Pin` interrupt handlers](http://docs.micropython.org/en/latest/library/machine.Pin.html?highlight=interrupt%20machine#machine.Pin.irq) and [`Timer`-based interrupt handlers](http://docs.micropython.org/en/latest/library/machine.Timer.html). The MicroPython documentation  [describes a number of restrictions, caveats and warnings about interrupt handler routines](http://docs.micropython.org/en/latest/reference/isr_rules.html). These kinds of issues, and the issues of using concurrency and shared objects safely, are some reasons we have felt interrupt support as found in MicroPython and Arduino is not a great match for our target audience.

What are your own specific cases for interrupts (that is, asynchronous event support)?
1. What do you need to trigger on? E.g.:
  a. A pin level change or edge.
  b. Some microcontroller internal event (which ones?).
  c.  A timer-based interval.
2. What kind of latency is OK? (E.g., "soft" vs "hard" interrupts)
3. What do you need to do when the event happens? E.g. (not inclusive):
  a.  Set a flag, to be checked later
  b.  Do something immediately (e.g., change another pin's value).
  c. Ask for something to be done in the near future (e.g. read a sensor, write out some data).

We would not expect to satisfy all the requirements people have, but studying the kinds of things people want to do will be helpful towards coming up with mechanisms. Thanks. (For instance, #4540 describes of a use case that I thought of while thinking about posting this issue.)

anecdata, 2021-04-02:
Primary use case for me is pin and timer interrupts that will preempt long or unpredictable-length code, like Requests.

ktritz, 2021-04-04:
1.) A trigger on a pin-level change from a Hall or other type of limit switch during position feedback control.

2.) The faster the response, the more accurate the positioning, regulated by actuator velocity of course

3) Stop the positioning

Both the digital input and the motor pwm is over I2C communication, so super-fast polling will interfere with other I2C communication activities.

microDev1, 2021-04-04:
Just coming in from a really nice discussion on discord about interrupts with a proposal.
Having `alarm` api return the number of times an alarm got triggered should solve most interrupt use cases.

The api can look like:
```python
pin_alarm = alarm.pin.PinAlarm()
alarm.count_on_alarm(pin_alarm)
triggered_counter = pin_alarm.count
```

This might render `countio` obsolete.

ladyada, 2021-04-04:
@ktritz this would be way better handled by countio thru direct connection to GPIO... no interpreted application will be good enough to read a hall effect sensor, over i2c, for motor positioning. i wouldn't even build a motor-control-over-i2c in C :/

ktritz, 2021-04-04:
I'm looking at using an I2C pwm breakout to drive the motor ESCs, and and I2C digital breakout to read the Hall sensor switches. Why? Because we need to MUX a lot of these, and I wanted to keep wire count down on the main MCU board, in addition to providing easier forward compatibility to changing the MCU (just have to worry about the I2C instead of matching all of the PWM/DIO pins). You are certainly right to be cautious that this has a lot of communications overhead, but I our time scales should be slow enough that this should be ok. However, lots of I2C polling for the hall switch condition could be reduced by taking advantage of the Interrupt line on the I2C DIO muxes, feeding this into countio and then polling that sounds like something to try.

WarriorOfWire, 2021-04-04:
> without glitching when there is a request or other long atomic operation.

Requests consist of (set_up, send, wait for a tremendous amount of cpu time, receive). If that waiting is undesirable, the right way to keep the application responsive is via an async/await request api. If you know how long you have until your next colon flash (and if you only have a blocking requests.get), you should just pass in that timeout value to the requests.get invocation right? This is one way that async/await really helps people make simple and responsive apps. See https://circuitpython.readthedocs.io/projects/requests/en/latest/api.html#adafruit_requests.Session 's request() method and its timeout parameter for ideas on how to accomplish this flashing colon today.

**On topic:**

There are a couple constraints imposed on any interrupts by CircuitPython philosophy that are driving my opinions here:
1. There is 1 Python stack.
1. CircuitPython code runs in a normal stack.
1. CircuitPython should be predictable for users. (pressure toward ease of use by beginners with less mystical behavior)

**What do you need to trigger on?**
- pin level change and hardware timer, as you noted.
- microcontroller event: Stream protocol "more bits arrived"
- microcontroller event: UART/I2C bits received
- microcontroller event: displayio frame scan out completed

**What kind of latency is OK?**
I think this should be driven by what works for CircuitPython more than by the question itself.

Now, philosophy 1 & 2 are subtly different, and if they hold true they together dictate that interrupts must be handled in normal, unrestricted user Python code. For 3, I think it means that it would be highly shocking to a user if simply by configuring an interrupt pin to count button presses suddenly all of the sleeps started raising exceptions.

Regardless, exceptions or any other means, by having the interrupts handled in a normal stack there is necessarily 10's, or more likely 100's of microseconds of latency at best. This is okay and just means that CircuitPython Python code might not be the right tool for 1us latency responses - which realistically shouldn't come as a shock to most users. I mean, you gotta remember garbage collections happen and that'll cost you 10's of millis at least!

**What do you need to do when the event happens?**
- execute a web request
- update a position of a displayio object
- send a UART message to an attached coprocessor
- send an I2C to read a timer
- send a request to reddit.com, read the json response, update e-ink display
- check if enough pulses have happened to update the logical position of a rotary encoder.
- infinitely many more things.

2 ways occur to me to satisfy the 3 philosophical constraints and provide interrupt-driven features for projects:
### Easy Way
A polling interface.
```
button_interrupt = interrupts.Interrupt('a button pressed!', pin1, FALLING)
button_release_interrupt = interrupts.Interrupt('a button lifted!', pin1, RISING)
button_interrupts = { button_interrupt.id: button_interrupt, button_release_interrupt.id, button_release_interrupt }

# First way to use polling: Always check all interrupts every loop
while True:
  press_count = button_interrupt.get_and_clear()
  release_count = button_release_interrupt.get_and_clear()
  for _ in range(press_count):
    print('button was pressed!')
  for _ in range(release_count):
    print('button was released!')

# Second way to use polling: Interrupt-driven activity for low power
while True:
  triggered_interrupt_ids = interrupts.wait_for_interrupt(max_wait=5)  # some upper bound on wait for liveness of any non-interrupt tasks
  for id in triggered_interrupt_ids if id in button_interrupts:
    for _ in range(button_interrupts[id].get_and_clear()):
      print(id)
```

### Hard Way
Dynamically branch out to interrupt handler callback code.

Here you'd have a potentially surprising call stack that would root at any random point and invoke your callback handler. This would be faster, but still tens-to-hundreds of microseconds' latency and harder to use correctly with enormous gotchas.

```
button_interrupt = interrupts.Interrupt('a button pressed!', pin1, FALLING, handler=lambda count: print(count))

[...]
# inside of requests.get
  read(buffer)
  for byte in buffer:
    b = handle_encoding()
    <------ WILD BUTTON APPEARED!
     ------ CircuitPython runtime invokes your handler _here_ instead of moving to the next line.
     ------ It is as though the interrupt.handler() lambda invocation was written here _one time only_.
     ------ When the handler returns this method continues as though it had called the handler itself.
     ------ But if this was on sleep() you'd have to resume sleep()... Not clear.
    output_buffer.append(b)
```

Hard way also forces thinking about how to handle a library sitting in sleep() which is not at all obvious what the expectation should be for users.

I mean this style is kind of neat and aligns well with my mental model of event-oriented programming, but on more careful consideration it's fraught with footguns, deceptively advanced and the stack is wicked. Also you don't know what a library method does internally. **E.g.,** I've written code that shuffles data from an SD card to a display's RAM as quickly as possible by orchestrating DMA transfers between 2 SPI buses. I allocate the largest 2 buffers I possibly can and manage them internally in the function. Having an interrupt fired and implicitly calling exte [... cut ...]
Either approach gives you power usage wins, but I think the Easy Way is easier to implement and __easier for users to use correctly__, while the Hard Way does not meaningfully reduce latency to the point that you could actually use CircuitPython Python code for microsecond-level ultra-precision timing execution.

Either approach lets you build advanced things like callbacks and interrupt-driven async/await coroutines. Hard Way might make this harder too, while making users' lives more difficult and unpredictable.

Letting users control their stack keeps interrupts predictable. Easy Way wins for me =)

Red-M, 2021-04-24:
Supporting interrupt pins from IO hooked up to other chips than the MCU or tasks that require to be action-ed every so often with timers.

Red-M, 2021-04-24:
Supporting interrupt pins from IO hooked up to other chips than the MCU or tasks that require to be action-ed every so often with timers.

EDIT: Being able to interrupt for a task that is latency critical is my main use case but could also provide a way to do concurrency without too much fuss.
Is part of this issue around concurrency education based?

daveythacher, 2021-04-25:
Lets back this conversation up and start over. Lets look at the fundamentals of the coding. Python, Java, etc. are mostly interpreted programming languages which normally reside in user space. Interrupts are normally reserved for kernel space. Many interpreted programming languages crutch off kernel APIs or use native code in the background.

This is mostly how CircuitPython works. It provides a large flexible API which works around the overhead. This works for bandwidth however really struggles with latency. The issue here is Python is still playing by control flow architecture. In theory Python could play by dataflow or transport triggered architecture. This is what these APIs in Python, Linux, etc. emulate.

Now I want to back up even further. Fundamentally you are asking about software model. In C on bare metal you have a couple different models: ISR, super loop, RTOS, etc. In Java/C++/etc. you have threading models: Executor Service, thread pool, runnable, async, coroutines, etc. Python however follows for the most part a simplified version of C.

CircuitPython lacks a kernel, scheduler, and ability to express native code. This cripples latency. The question here is not really about ISR's performance or implementation. Instead it is about the programming model. Python conceptually in this model is supposed to be glue logic. Its expression of logic is not supposed to be executed. The configuration component is supposed to be executed.

This design model is used in other places. (ASIC, SoC, MCU, etc.) However event systems can be tricky here. Latency of non-expressed logic will rely on interpreter which will be slow. Design of system will require module or other design section to handle/mitigate this. 

This however is an advanced topic. We could have this exact same conversion with Linux user space interface. This really is an interface question. Problem with interfaces is if they are too ridge they are useless. Too flexible they are bloated. Again this concept of circuitPython is known because Java has conceptually done this for decades.

However the conversion aspect of the front end has never really evolved. Abstraction of IO has never happened either. PIO in RP2040 may be able to get that conversation going. Perhaps circuitPython can get that conversation going. Hardware/Software Co-design is not a new field. However it has never been fully viable due to limited abstraction tools and hardware cost/availability. Time would eventually solve this. If nothing else bailing out Java could would create some kind of need for larger chip production.

Java, Linux, circuitPython, etc. is a productivity argument. Which is an economics argument. However it is not being used for these purposes completely. In summary it is important to consider what circuitPython is and what it is not.

ladyada, 2021-04-25:
hi folks, this issue is for collecting _specific_ use cases for interrupts so we can implement tools to help _solve specific problems_. its fun to talk about operating system design, please use discord for that so we can keep this thread on topic.

daveythacher, 2021-04-25:
Make code maintainable? Make code understandable? Make code not require insane testing? Is that not like the point of application code, APIs, libraries, etc.?

Anyone write state machine for polling in super loop?

Logic need low priority long running computation while fully saturating IO bus. - Maybe supported.
Logic needs to be target from other device. - Maybe supported.
Cannot redesign protocol or timing standard. - Maybe supported.

On reliability, can I freeze my code? Convert the dynamic nature to static nature. I know library which more likely to be dynamic however I should still be able to convert them to static at least in some part.

ThomasAtBBTF, 2021-04-25:
I would also like to look at the problem from a different perspective.
I agree very much with the statements of David Thatcher.
But my approach to the problem which we want to solve with introducing interrupts would go in a different direction. In general CurcuitPython is fast enough on most microcontroller to respond to Hardware events and do the processing in an adequate time. However longer running pieces of code are often in the way. Here I am helping with calling out to hardware event checking routines every so often in those longer running loops.
Activating multithreading support from MicroPython / CPython would help here enormously.
Users could easily move their long running code to a thread and could handle the hardware events in a main loop with suitable sleep() statement. While sleeping the threads can run “cooperative” and the response time to hardware events would be relatively predictable.
So how about giving up the idea of hardware interrupts in favor to multithreading.

dhalbert, 2021-04-25:
Enabling multithreading in general could be discussed in another issue. Feel free to open one. In this issue we are asking for _specific_ uses for that need handling of asynchronous events. I.e., what is the project you want to accomplish that needs such a capability?

daveythacher, 2021-04-26:
Go ahead and just mark this one as off topic also.

Interrupts are a couple of things. 
- They interrupt long running code. Making code structure simpler.
- They handle events in a non predictable manner as the source is "random" without overhead. (Ex. User input.)
- They mark critical events, which must be handled. Drop everything and do this. (Assuming processor can handle it.)
- They can be IPC or signal state alignment. Block till this or notify me which section you are in. (Ex. State machine, etc.)

Asynchronous in sequential logic denotes deferent things. The events trigger state change, at the event speed. This is not fundamentally possible in a processor. It is a synchronous sequential circuit emulating an asynchronous model using really slow state machine.  This is a huge topic.

Dictionary.com:
of or relating to operation without the use of fixed time intervals. (opposed to synchronous).

Other definition is:
of or requiring a form of computer control timing protocol in which a specific operation beings upon receipt or indication (signal) that the preceding operation has been completed.

Bringing it around to the point:

circuitPython is trying to assert that they will provide the real time information to you. They do not care about the code complexity. (Limitations of implementation, which I tried to mention earlier.) There are APIs which handle everything below the covers faster. They have interfaces which will create ISRs for IO reload. They have interfaces for counting pulses. They have interfaces for recording time between events with count.

I poll these APIs in a super loop which I divide up, since coherency uses too many resources in the common case on MCU. Using these interfaces there is no need to fundamentally provide user interrupts. However I lose real time control and code becomes complex notion of super loop. Python's latency and dynamic memory are fair points which negate most of this. 

However the amount of tutorial's required to explain this could be large. Is discord the best place? Are forms or blogs the best place? How will people find them? How many will read them? How good they will be? 

Technically there is a conflict in the theory in terms of simplification of expression. How much information do I need to understand about that which I express? Do I need to study module to learn who to apply it for this task? Or do I need to understand every aspect of the intended logic in terms of Input, Output and function?

This is kind of the problem with Java. Life is simple with Java, but a total pain in the rear at the same time. Java is sometimes called a sand box. Anything of sand with this amount of sand. For this limitation I get even more limitations. But I can do it really fast and maybe know nothing.

For beginners and stuff you just want fast and/or cheap this is may be great. So long as this is understood and learning is not hindered. I am not actually trying to bomb everything, but to explain what this is and what it is not. While expressing what it could be.

**In summary they want me to say something like I need FT232H (Adafruit sells them) with 245 FIFO asynchronous mode support on RP2040 with at least 3-8MBps. So that they can point me to a library or build me a library.  Why? To exceed 12Mbps. No USB protocol processing.** 

For studying look into working with sync version, which requires 60MHz synced IO. FPGA conversion could support it. 

This is likely outside scoping limits, but FPGA timing extension could be possible with circuitPython. Good luck explaining that. Whole other order of expression and interface topic. Basically a retrofit which will have scope issues. Trying to work out timing also. People tried something like this for Arduino I think. They did not take it very far.

The IO bus for controllers is a fun topic. However this is also known to potentially work back up into hardware software codesign which would potentially work for processors. Another fun topic. Most abstraction is a joke.

This model will work, however it will not work for everything. This will likely never be used in medical, military, etc. environments. However for maker projects it will enable a decent amount of stuff with little to no real problem. Lot of entropy is inferred here, which in theory could be removed. However this is not that and that has been on the horizon for a long long time.

daveythacher, 2021-04-26:
In my example I did not provide any internal details. Because I will not be implementing those. So instead I just told you my requirements. You will figure it out.

This notion of soft vs hard, internal vs external, etc. is really off topic from what you really appear to be asking for. Therefore I reclassify the topic back to fundamental reasons for interrupts.

daveythacher, 2021-04-27:
I skipped over polling interrupts and ISR details. I am not going to go through the exercise of explaining all of that just to contrast it against async. Which will mostly be a thread model, which was marked as off topic. If you have the right hardware you can build a RTOS in hardware. (I am not aware of that existing.) Conceptually interrupts can be modeled as async however this is technically correct and technically wrong depending on context.

Linux supports kernel modules for a reason. circuitPython does not support this or a scheduler.  circuitPython has static modules built into the main interpreter.  However some dynamic support may exist, in unsupported and hackish manner. circuitPython is trying to avoid coherency challenges. So I am not sure I really see where this is going.

Technically the issue is getting off topic. Async is technically a thread model, but maybe more lightweight. Conceptually you have an execution problem. Getting processor time and how to express it for performance. Then there is coherency of state and overhead considerations in multiple manners.

I do not believe this to be a fallacy initially however I suspect it has become this. I would recommend this to be closed. Maybe reopen under more clear intentions.

jposada202020, 2021-04-28:
Adding here the reference on issue https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_Layout/issues/19#issuecomment-828394272
Basic use case is some async event to better respond to user interactions. Could be actions on the reactive screens or other depending on user design.  
The Event loop is the main discussion here and how to verify different states of differents graphical components that can be used at the same time producing different results and actions in the code.

dglaude, 2021-05-05:
Playing with Thermal display, you can be stuck into a function call to the library for a "long time" and you cannot catch event (button or else) while receiving your frame.

This usecase is discussed at the end of this blog post: https://www.recantha.co.uk/blog/?page_id=20924

jweaver85, 2021-05-16:
Greetings!

Short background:
I've spent a fair amount of time attempting to understand how my Raspberry Pi Pico works.... I've barely scratched the surface. Thus far I've been able to get some "neopixels" running some basic algorithms, such as a walk from (x1,y1,z1) to (x2,y2,z2) these can be pre-determined coordinates or randomly generated to create really fun color effects. One of my implementations uses a typical Raspberry Pi running a python daemon consuming Firebase events and updating brightness, sleep time, step size, and the color effect to run. 

Use case:
I'm porting this functionality and using buttons to replace the Firebase portion. Without the ability to use Timer + callback, GPIO initiated interrupts, or threading I can only check the status of the button once every sleep interval which is consequently is also directly tied to the rendering of the "neopixels." Here my "code.py" which should demonstrate this: 
```
import time
import board
import neopixel
import digitalio
from deque import deque
from queue import queue
from rainbowwalk import rainbowwalk

# Update this to match the number of NeoPixel LEDs connected to your board.
num_pixels = 60
colors = queue([], num_pixels)

# options
step = 1
brightness = 0.1
sleepytime = 0.1
algos = ["rainbowwalk", "off"]
algo_index = 0

pixels = neopixel.NeoPixel(board.GP0, num_pixels)
pixels.brightness = brightness
pixels.auto_write = False;

algo_button = digitalio.DigitalInOut(board.GP13)
algo_button.switch_to_input(pull=digitalio.Pull.DOWN)

brightness_button = digitalio.DigitalInOut(board.GP12)
brightness_button.switch_to_input(pull=digitalio.Pull.DOWN)

def updateAlgorithm():
    global algo_button
    global algo_index
    global algos
    if algo_button.value:
        algo_index = (algo_index + 1) % len(algos)
        print("algo_button pressed! Current algo: " + str(algos[algo_index]))

def updateBrightness():
    global brightness_button
    global brightness
    global pixels
    if brightness_button.value:
        if brightness + .1 > 1.0:
            brightness = .1
        else:
            brightness = brightness + .1
        pixels.brightness = brightness 
        print("brightness_button pressed! Current brightness: " + str(brightness))
        
    
while True:
    updateAlgorithm()
    updateBrightness()

    if algos[algo_index] == "rainbowwalk":
        rainbowwalk(step, brightness, sleepytime, pixels, colors)
    
    if algos[algo_index] == "off":
        pixels.fill((0,0,0))
        pixels.write()
        
    time.sleep(sleepytime) # <-- PRIMARY LIMITATION OF IMPL
```

As you can see, I am extremely limited on adjusting "sleepytime" which I eventually hope to attach to a button also. If the value is too small, it will register the button push several times, and if to slow, the user will have to hold down the button for an unintuitive period. Although I enjoy the happy side effect of holding down the algo_button, which has a strobe light effect, it is undesired.

In closing:

1. What do you need to trigger on?
    - I can trigger on anything. Whatever is easiest to implement for you folks 
2. What kind of latency is OK? (E.g., "soft" vs "hard" interrupts)
    - Ideally "hard," however for a sufficiently high core frequency, "soft" shouldn't be a problem
3. What do you need to do when the event happens? E.g. (not inclusive):
    - Do something immediately (increment/decrement values)

Thanks for making awesome stuff!

WarriorOfWire, 2021-05-17:
Hi @jweaver85, I think you can accomplish what you're trying to do with Tasko or your own event loop & circuitpython's async/await support today. Should look something like:

```
import tasko

[...]

# Scheduled at a high rate so that the button is responsive
asyc def updateAlgorithm():
    [unchanged, maybe have one of these per animation instead of spreading them across time? or loop them in this function]

# Scheduled at high rate to keep brightness button responsive
async def updateBrightnessCoroutine():
    [unchanged]

async def do_rainbowwalk():
  [get step, brightness, sleepytime i guess?, pixels, colors]
  rainbowwalk(step, brightness, sleepytime, pixels, colors)

def setup():
    tasko.schedule(hz=20, coroutine_function=updateAlgorithm)
    tasko.schedule(hz=20, coroutine_function=updateBrightness)
    tasko.schedule(hz=100, coroutine_function=do_rainbowwalk)

if __name__ == '__main__':
  setup()
  tasko.run()
```

Note that cooperative multitasking (ez mode) and interrupts/preemptive simultaneity (hard mode) are separate concepts =)

jweaver85, 2021-05-18:
@WarriorOfWire 

Thanks for this suggestion! I was able to get the async functions running (almost identical to the code you provided). Part of my issue was self inflicted, I wasn't keeping track of the previous state of the button (ie, long press causing a strobe light effect because there were only two ... ['rainbowwalk', 'off']) 

The async polling works great!

guljeny, 2021-07-12:
I'm doing keyboard with display. On display showing total pressed keys, I'm need to write this in controller memory (I'm using pi pico), but all memory operations so long (beetwen 0.4 and 0.7 sec for write). If I press key while data writing, I miss this key (or show with really big delay). My controller has two cores, but now it impossible to use both of them or write data in async operation which no lock main cycle

tannewt, 2021-07-12:
> I'm doing keyboard with display. On display showing total pressed keys, I'm need to write this in controller memory (I'm using pi pico), but all memory operations so long (beetwen 0.4 and 0.7 sec [...]

async won't help you on the RP2040 because the code is running from the same flash you are trying to write to. Core count doesn't matter. We can't load code off the flash while a write is happening and we must block as a result. Your best bet is having a separate non-volatile memory that isn't shared with storing code. [Here](https://www.adafruit.com/product/1897) is an example.

guljeny, 2021-07-13:
@tannewt yea, I was solved this by using external EEPROM. But so many cases in my project when I'd like to use threads. Two  text lines on display (len. around 14 sym), each of them updating around 40-60ms, and together in locks my keyboard on 100ms (it's really long when typing). Solved by slice lines and update each symbol separately at different time. To many math (total around 120ms) - solved by partially execute all unimportant functions in loop, step by step. Of course, all problems solved, but today more and more controleer use 2+ core cpu, and better way use this =(

lutzray, 2021-07-17:
I'm driving the SAMD21 DAC to generate a pulse going into an audio recorder AUX channel for video/multicam/audio syncing. Yes, this is _Yet another linear time code_ (YaLTC for short). The pulse is triggered by a GPS PPS. For now I'm using an ISR within the Arduino IDE with trinket, QT-Py and Feather M0 proto boards for my prototypes (with  your Quectel L80 breakout/wing boards). I know I could have used the PPS signal itself, fed into the audio recorder input but I want to keep the IC count low. I wish I could have used CircuitPython for this (but on a brighter side, it forced me to clean up my rusted C skilz). See my [hackaday project page
](https://hackaday.io/project/176196-atomic-synchronator)

mudddog2002, 2021-07-18:
Hi!  I'm a hobbyist who enjoys making stuff.  I have an EE degree, but have never really programmed consistently to make progress past the basics (i.e. I have to look things up from scratch every time).  That said, I have some basic understanding of what I'm trying to accomplish, but often struggle to find an efficient way to accomplish it.

I usually find myself thinking in state machines.  It helps me logically decide what to do and to stay organized.  I think a good interrupt use case is for a button to advance the state machine to the next state.  For example, today I'm playing with a DotStar LED strip.  I'd like to have a few different modes (POV Painter, Light Show, and Interactive).  I'm struggling to find the right point in the POV painter loop that will be responsive enough to trigger on a button press, but not too zippy that it jumps states.  If there was an easy way to "Advance State Machine Counter on Button Press", then I could just check the state very often and life would be good.  It would also be helpful to use this as a "low power wakeup"...state 0 could be basically asleep.  When the button is pressed, go to state 1.  

I'm sure many of these things can be implemented, but there seems to be a big leap in code complexity once we get past the basics (I can easily blink a light, check a button, read a sensor, etc.  But combining them to do something interesting always leads to a lot of complexity with weird code functions that I don't know how to use.  Then I end up wasting hours going down google rabbit holes).  So...in other words, help us LadyAda...a few basic asynchronous options that keep the projects humming along would be great!

PS I ran out to MicroCenter yesterday to get a CircuitPython Feather Express.  It was awesome that I could have it in my hand in an hour!  But the parts are hard to find on the shelf (lots of very small labels on bags that look identical, and a lot of products jumbled up together).  Better than nothing for sure, but I still dream of a beautiful sliding drawer organizer full of Adafruit components labeled and organized nicely like Radioshack used to have for resistors and ICs.

tannewt, 2021-07-20:
> I'm driving the SAMD21 DAC to generate a pulse going into an audio recorder AUX channel for video/multicam/audio syncing. Yes, this is _Yet another linear time code_ (YaLTC for short). The pulse is [...]

What would the Python code do? Your project says 10ns accuracy and I'd worry that any Python level code would move the accuracy into the 10s of us at minimum.

lutzray, 2021-07-20:
10 ns is the PPS GPS spec, the actual sync precision (in postprocessing the wav file on the desktop) is more or less an audio sample, eg 20 μs for 48kHz sampling (as illustrated in this [project log](https://hackaday.io/project/176196-atomic-synchronator/log/194541-almost-a-kickstarter-sales-pitch). My python code would do Direct Digital Synthesis of a 2FSK word encoding 24 bits of UTC time.

NikilXYZ, 2021-08-18:
> Regardless, exceptions or any other means, by having the interrupts handled in a normal stack there is necessarily 10's, or more likely 100's of microseconds of latency at best. This is okay and [...]

Pretty sure that interrupt context switching on AVR microcontrollers took around 40 uS anyway. So this is within reasonable expectations.

WarriorOfWire, 2021-08-18:
How many microseconds does python gc take on your AVR though? That’s the real point there, not interrupt context switch timing (both of which are different from circuitpython loop lifecycle state- please feel free to read the tick loop code if you’re curious, and gather your own data on gc timing if you doubt mine).

Circuitpython is not the right tool for microsecond timing critical projects, and they’d be out of scope for Circuitpython as well as this async use case issue! You’re not going to run a Hall effect feedback loop brushless motor reliably, for example - at least not at any moderately high rpm - with any kind of python code async or not. You’ll glitch out at every gc event _at least._

focus on tasks that involve waiting for something. Like a BME680, where you have to heat the air sensor, wait, send a request to sample, wait, read the reply. Those wait points are great opportunities to go update your displayio widgets and resume after some time and if your timing is off by a few milliseconds it’s no big deal. The messages just waited in a buffer briefly.

obvs you could use an interrupt to signal that wait time is over but that really doesn’t change the shape of the problem or the solution very much!

NikilXYZ, 2021-08-18:
Well there's no Python on AVR as far as I know.

I am just saying ppl used to AVRs through arduino unos and subsequent
boards might not have an issue with that timing

waseltzer, 2021-11-15:
Disappointed to learn that CircuitPython doesn't support interrupts. I'm working on curriculum to teach *Python on the Pico for a college class and interrupts are an important concept. I'll have to use MicroPython for that exercise.
Like this:

`import machine
import time

from machine import Pin

led = Pin(25,Pin.OUT) #pico on-bard LED
button = Pin(20,Pin.IN,Pin.PULL_UP) #button

def button_int_handler(pin):
    led.toggle()

led.off()

#call the interrupt handle when the button is pressed OR released; the LED is on when button is pressed, off when released
button.irq(trigger=(button.IRQ_FALLING | button.IRQ_RISING), handler=button_int_handler)
`

dhalbert, 2021-11-15:
> I'm working on curriculum to teach *Python on the Pico for a college class and interrupts are an important concept. I'll have to use MicroPython for that exercise.

Note that MicroPython has strong restrictions on what you can do in a hard-interrupt handler: no storage allocation, etc. We are avoiding hard-interrupt handlers in CircuitPython to make it harder to make those kinds of mistakes. Functionality that needs hard interrupts we code as native modules. `keypad` does not actually use pin interrupts: it polls the pins based on the given interval. There is an internal 1msec timer tick interrupt that schedules these.  `countio` does use hard interrupts. But you still have to poll. But see next comment for an update on where we are on this.

dhalbert, 2021-11-15:
Progress report on all this: we enabled `async/await` some time ago on most CircuitPython builds, and are now providing a preliminary port of MicroPython's `asyncio` implementation so you can do **cooperative multi-tasking**. This is in 7.1.0-beta.0 and later. A preliminary Learn Guide with examples will be available soon.

We chose `asyncio` because it's standard in CPython and MicroPython already has done a great job on the implementation. However, `asyncio` is not so easy to use or understand, and it's relatively easy to make mistakes that can bite you later, particularly with respect to error handling and task cancellation. [`curio`](https://curio.readthedocs.io/en/latest/) and [`trio`](https://trio.readthedocs.io/en/latest/) are better in this regard, because they provide more "structured concurrency" mechanisms, particularly task groups (aka "nurseries"), and do not have as many kinds of task-like objects (Futures, Tasks, and co-routines are all visible in asyncio). We looked at porting both `curio` and `trio`, but it would have been a substantial job.

Originally, the `asyncio` folks thought they could get task groups into Python 3.8. However, handling multiple exceptions at once was a stumbling block, and it has taken until Python 3.11 to get language features in to handle them.

There are some simple task-group implementations for `asyncio` in [`aiotools`](https://aiotools.readthedocs.io/en/latest/) and [`anyio`](https://anyio.readthedocs.io/en/latest/), a library that wraps `trio` or `asyncio` with a common API.

There are also some simpler cooperative multitasking libraries already available for CircuitPython
- [`asynccp`](https://github.com/WarriorOfWire/CircuitPython_async)
- [`meanwhile`](https://github.com/deshipu/meanwhile) (an existence proof that does not use `async/await`)

Cooperative multitasking provides an easy way to do structured polling, without having to litter your code with polling calls. In many cases where interrupts would otherwise be used explicitly, it may be enough. We are also going to look at how provide do async versions of various built-in modules, such as the `busio` communications protocols.

dhalbert, 2021-11-24:
The Learn Guide [Cooperative Multitasking in CircuitPython](https://learn.adafruit.com/cooperative-multitasking-in-circuitpython) has been published. More parts need to be added to it, but it currently includes a number of simple examples.

I'm closing this issue for now. Thanks for all your input! See the previous comment for more information. Please open other issues for asyncio followups and suggestions for further enhancements.

closed 2021-11-24 by dhalbert