adafruit/circuitpython #4546: Support for micro:bit v2 (nRF52833)

tyomitch, 2021-04-03:
Following the guide at https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython/ :

* copied `boards/simmel`, which is another nRF52833-based board;
* updated `mpconfigboard.mk` with
```
USB_VID = 0x0d28
USB_PID = 0x0204
USB_PRODUCT = "micro:bit"
USB_MANUFACTURER = "BBC"
```
* updated `mpconfigboard.h` with
```
#define MICROPY_HW_BOARD_NAME       "micro:bit"

#define DEFAULT_I2C_BUS_SCL         (&pin_P0_26)
#define DEFAULT_I2C_BUS_SDA         (&pin_P1_00)

#define DEFAULT_UART_BUS_RX     (&pin_P0_06)
#define DEFAULT_UART_BUS_TX     (&pin_P1_08)
```

According to the guide, this should have been enough to get the REPL running over the serial interface. The connection to the serial interface succeeds, but in response to any input, I just see the amber status LED flashing, but no output.

What am I missing?

The pinmap (and other board data) are published at https://tech.microbit.org/hardware/schematic/

It would seem that CircuitPython had supported the original (nRF51822) micro:bit until 2018 when the support for all nRF51 boards was dropped (05bd5d4; #591)

dhalbert, 2021-04-03:
The nRF chip on the micro:bits is not connected directly to the USB pins; it goes through the interface unit microcontroller, as you may know . So I'm not surprised you are not seeing the REPL. All the current CircuitPython code assumes that the REPL is over USB. We never supported the micro:bit V1 with CircuitPython. There is a specialized port of MicroPython that did a lot of squeezing to get it to fit in the small RAM and flash space.

Currently, we would not support the micro:bit V2 because there is no CIRCUITPY experience possible due to the lack of a mass storage USB connection. We have started working on an alternative workflow experience that could be BLE-only, and it's conceivable that the micro:bit V2 could be supported in that way.

tyomitch, 2021-04-03:
> The nRF chip on the micro:bits is not connected directly to the USB pins; it goes through the interface unit microcontroller, as you may know . So I'm not surprised you are not seeing the REPL. All the current CircuitPython code assumes that the REPL is over USB. 

Yes, but the interface controller is supposed to expose the specific pair of nRF pins (P0.06 and P1.08) as a USB serial interface -- that's my understanding from their schematic page. Mass storage experience aside, why does the non-direct USB connection stop the REPL from working?

(Btw, v2 has 128 KB RAM, so would no longer need much squeezing)

dhalbert, 2021-04-03:
> why does the non-direct USB connection stop the REPL from working?

The CircuitPython REPL code is sends and receives characters over USB serial port using TinyUSB, not via the UART port you mentioned (P0.06 and P1.08). There is some latent support for this in the `nrf` port, but we haven't tried it out in a while and it may have gone stale. Try setting `CIRCUITPY_REPL_UART = 1` in `mpconfigboard.mk`, which primarily controls code in `ports/nrf/supervisor/serial.c`.

dhalbert, 2021-04-03:
[note edit in previous comment]

tannewt, 2021-04-05:
I think the micro:bit v2 is a prime candidate for BLE peripheral only CircuitPython. I wonder if we could squeeze into a v1 as well. The 16k RAM is definitely the toughest part.

tyomitch, 2021-05-03:
@dhalbert It appears that my problem arises long before the serial connection; even if I add some blinking into `port_init()`, it doesn't run in my microbit_v2 build.

Looking in my `firmware.elf.map`, I see:
```
Memory Configuration

Name             Origin             Length             Attributes
FLASH            0x0000000000000000 0x0000000000080000 xr
FLASH_MBR        0x0000000000000000 0x0000000000001000 xr
FLASH_SD         0x0000000000001000 0x0000000000026000 xr
FLASH_ISR        0x0000000000027000 0x0000000000001000 xr
FLASH_FIRMWARE   0x0000000000028000 0x000000000003c000 xr
FLASH_BLE_CONFIG 0x0000000000064000 0x0000000000003000 r
FLASH_NVM        0x0000000000067000 0x0000000000000000 r
FLASH_FATFS      0x0000000000067000 0x0000000000013000 r
FLASH_BOOTLOADER 0x000000000007a000 0x0000000000004000 xr
FLASH_BOOTLOADER_SETTINGS 0x000000000007f000 0x0000000000001000 r
RAM              0x0000000020000000 0x0000000000020000 xrw
SD_RAM           0x0000000020000000 0x0000000000008000 rw
SPIM3_RAM        0x0000000020008000 0x0000000000000000 rw
APP_RAM          0x0000000020008000 0x0000000000018000 xrw
*default*        0x0000000000000000 0xffffffffffffffff

Linker script and memory map

                0x0000000000000000                _minimum_heap_size = 0x0
                0x0000000020020000                _estack = (ORIGIN (APP_RAM) + LENGTH (APP_RAM))
                0x0000000020020000                _ram_end = (ORIGIN (APP_RAM) + LENGTH (APP_RAM))
                0x0000000020020000                _heap_end = 0x20020000
                0x0000000020008000                _spim3_ram = ORIGIN (SPIM3_RAM)
                0x0000000020008000                _spim3_ram_end = (ORIGIN (SPIM3_RAM) + LENGTH (SPIM3_RAM))

.isr_vector     0x0000000000027000      0x100
                0x0000000000027000                . = ALIGN (0x4)
 *(.isr_vector)
 .isr_vector    0x0000000000027000      0x100 firmware.elf.lto.o
                0x0000000000027000                __Vectors
                0x0000000000027100                . = ALIGN (0x4)

.text           0x0000000000028000    0x3a258
                0x0000000000028000                . = ALIGN (0x4)
 *(.text)
 .text          0x0000000000028000       0x2c build-microbit_v2/supervisor/cpu.o
                0x0000000000028000                cpu_get_regs_and_sp
 .text          0x000000000002802c      0x378 /home/tyomitch/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/thumb/v7e-m+fp/hard/libgcc.a(_arm_addsubdf3.o)
...
```
-- i.e. nothing is loaded into `FLASH_MBR` or `FLASH_SD` regions, and `FLASH_ISR` which includes `__Vectors` which point to `Reset_Handler` among others, is loaded at 0x27000, so it's unsurprising that `Reset_Handler` doesn't run.

Now, `ports/nrf/mpconfigport.h` has:
```
// Flash layout, starting at 0x00000000
//
// - SoftDevice
// - ISR
// - firmware
// - BLE config (bonding info, etc.) (optional)
// - microcontroller.nvm (optional)
// - internal CIRCUITPY flash filesystem (optional)
//   The flash filesystem is adjacent to the bootloader, so that its location will not change even if
//   other regions change in size.
// - bootloader (note the MBR at 0x0 redirects to the bootloader here, in high flash)
// - bootloader settings

// Define these regions starting up from the bottom of flash:

#define MBR_START_ADDR  (0x0)
// MBR_SIZE is from nrf_mbr.h
#define SD_FLASH_START_ADDR   (MBR_START_ADDR + MBR_SIZE)

// SD_FLASH_SIZE is from nrf_sdm.h
#define ISR_START_ADDR  (SD_FLASH_START_ADDR + SD_FLASH_SIZE)
#define ISR_SIZE        (4 * 1024)   // 4kiB
```
-- so the memory layout I'm getting is exactly as intended for nRF52833. (I also tried to `#define ISR_START_ADDR  (0)`, but that didn't help either.) Yet, I don't see anything in the makefiles which would bring the actual MBR and SD code into the firmware binary; `$(SOFTDEV_HEX)` is used only with the flashing tools, which I don't have or use, preferring to just copy `firmware.hex` over USB.

What am I missing this time?

PR #5002 (2021-07-15): Create first BLE-only board, Micro:Bit v2
This fixes build issues with USB off, tweaks the README to allow
for BLE-only boards and adds the Micro:Bit v2 definition.

Fixes #4546 Fixes #4680
commit 0fadf028 (2021-07-15): Create first BLE-only board, Micro:Bit v2

This fixes build issues with USB off, tweaks the README to allow
for BLE-only boards and adds the Micro:Bit v2 definition.

Fixes #4546
closed 2021-07-15 by jepler