adafruit/circuitpython #4551: Linux not able to build CircuitPython

daveythacher, 2021-04-04:
Not able to compile circuitPython beyond 6.1.0 using Linux. Master and 6.2 do not work using Linux.

I tried the 6.2 RC and it does not work. Seen before in #4479

dhalbert, 2021-04-04:
You need to install some new prerequisites and bring in a new submodule or two to build. Ask on discord or in the forums for interactive help for this. The Building CircuitPython Learn Guide has been updated. See both the Linux setup part and also on the Build page, a new requirement of doing `pip3 install -r requirements-dev.txt`.

closed 2021-04-04 by dhalbert

daveythacher, 2021-04-04:
> You need to install some new prerequisites and bring in a new submodule or two to build. Ask on discord or in the forums for interactive help for this. The Building CircuitPython Learn Guide has [...]

I followed the build steps...so not sure I see your point. I installed the recommended compiler. I sync and initialized the submodules. If this is true it is not documented. Maybe you could point me in the right direction first? This issue will work as documentation, since I search for the issue an found 4479. You think?

I do not understand why this was closed. Redirecting to discord is wrong on many levels.

dhalbert, 2021-04-04:
We don't use GitHub for support questions like this. Please ask in discord or the forums. I am happy to help you right now in discord; I need more details like your error messages.. It is possible to build on Linux; that's what the core devs do all the time.