adafruit/circuitpython #4552: Web interface for custom firmware builds

microDev1, 2021-04-04:
Proposal for having a web interface to build custom firmware where only the modules that are required for a specific project can be selected and if storage space is saved doing this it can be added to the `CIRCUITPY` drive.

Something like this: https://nodemcu-build.com/

tannewt, 2021-04-05:
I'm ok with this but would ask that it *not* label the builds with the CircuitPython name. Creating essentially infinite combinations will be a huge support burden. (We have trouble with folks mixing different CP versions already.)

ajs256, 2021-04-06:
> that it not label the builds with the CircuitPython name

How about "CustomPython"? Or "MyPython"?

tannewt, 2021-04-06:
CustomPython would work MyPython might be too close to [MyPy](http://mypy-lang.org/).

hierophect, 2021-04-06:
SockPython? That way we can make one of these for Blinka:
[image]

MicroDev1, 2023-02-17:
You can now make custom board builds in your fork. See https://github.com/adafruit/circuitpython/pull/7594#issuecomment-1432525280.

closed 2023-02-17 by tannewt