adafruit/circuitpython #4562: main ulab guide: modernize when 7.0 goes out

jepler, 2021-04-05:
We updated ulab and it moves most items around, e.g., `ulab.linspace` becomes `ulab.numpy.linspace`.  Update the guide, but only when 7.0 is out or is at least very imminent.

jepler, 2021-08-24:
@FoamyGuy is this something you ended up taking on or does it still need to be done?

FoamyGuy, 2021-08-25:
@jepler I did make these changes in the learn guide at the time that the PR in the Learn Repo was merged.

I believe I got everything updated that needs it. But I am less experienced with ulab (or numpy / scipy) so it may be worth taking a quick look over it to ensure I didn't get anything wrong if you or someone with more experience is up to doing that.

jepler, 2021-08-25:
I'll skim the guide text before closing this up.

jepler, 2021-08-25:
I fixed up some text on the last page, thanks for doing the bulk of the work @FoamyGuy !

closed 2021-08-25 by jepler