adafruit/circuitpython #4563: reset intermittently triggers "Crash into the HardFault_Handler" on MagTag

mlupo, 2021-04-05:
Using `microcontroller.reset()` or `exit_and_deep_sleep()` intermittently triggers a "Crash into the HardFault_Handler" and reload into safe mode on the MagTag using CircuitPython 6.1.0.

I encountered this issue while making a simplified derivative of the [MagTag cat fed clock](https://learn.adafruit.com/magtag-cat-feeder-clock) project, in which I used `exit_and_deep_sleep()` to sleep the MagTag in-between minute refreshes. With `exit_and_deep_sleep()`, the MagTag would hard crash after around 9 hours of continuous operation. After a manual reset, the device would operate as normal on the same battery charge for another ~9 hours before crashing. In trying to to debug the issue I replaced `exit_and_deep_sleep()` with a simple `time.sleep()` then `microcontroller.reset()`. Unfortunately, this version would hard crash in ~3 minutes.

I am using a ChronoDot for time keeping, but as far as I can tell the same crash occurs when I use Adafruit IO over wifi. Full code is:

```python
from adafruit_magtag.magtag import MagTag
from busio import I2C
import adafruit_ds3231
from board import SCL, SDA

CLOCK_I2C = I2C(SCL, SDA)
rtc = adafruit_ds3231.DS3231(CLOCK_I2C)

magtag = MagTag()
magtag.display.rotation = 90
magtag.peripherals.neopixel_disable = True
magtag.peripherals.speaker_disable = True

magtag.graphics.set_background(0x000000)

mid_x = (magtag.graphics.display.width // 2) - 1
magtag.add_text(
    text_font="Lato-Regular-74.bdf",
    text_position=(mid_x, 10),
    text_anchor_point=(0.5, 0),
    is_data=False)

magtag.add_text(
    text_font="/BebasNeueRegular-41.bdf",
    text_position=(mid_x, 86),
    text_anchor_point=(0.5, 0),
    is_data=False)

clock = rtc.datetime

weekdays = ("mon", "tue", "wed", "thur", "fri", "sat", "sun")
months = ("Octember", "January", "February", "March", "April", "May", "June",
          "July", "August", "September", "October", "November", "December")

hour = clock.tm_hour
minute = clock.tm_min
weekday = weekdays[clock.tm_wday]
month = months[clock.tm_mon]
day = clock.tm_mday
year = clock.tm_year

if hour > 12:
    hour = hour - 12

if minute < 10:
    minute = "0" + str(minute)

magtag.set_text(f"{hour}:{minute}", auto_refresh=False)
magtag.set_text_color(16777215, index=0)
magtag.set_text(f"{weekday}. {month} {day}", index=1, auto_refresh=False)
magtag.set_text_color(16777215, index=1)

magtag.refresh()

clock = rtc.datetime
wait = 62 - clock.tm_sec

magtag.exit_and_deep_sleep(wait)
```

If it matters any, I am using macOS Big Sur and sending the main.py file with the cp -X command.

Thank you so much for CircuitPython and all the work that goes into it!

tannewt, 2021-04-05:
Please try 6.2.0-rc.0 (or 6.2.0 if it's released.) We've updated a lot of stuff since then.

mlupo, 2021-04-05:
Thank you for the reply, and good news so far: `microcontroller.reset()` no longer triggers a crash with 6.2.0-rc.0. I will have to let the device run for quite a bit in order to test if `exit_and_deep_sleep()` still malfunctions, but things are looking good so far.

tannewt, 2021-04-07:
Thanks for following up @mlupo. No worries having it open while you investigate.