adafruit/circuitpython #4574: Teensy 4.0/4.1 hardware RTC does not keep time unpowered while having vbat supply

Remowylliams, 2021-04-08:
**Firmware**

```python
Adafruit CircuitPython 6.2 and earlier on 2021-04-07; Teensy 4.0/4.1
```

**Code/REPL**

```python
"
import rtc
import time

r = rtc.RTC()
r.datetime = time.struct_time((2019, 5, 29, 15, 14, 15, 0, -1, -1))
"
```

**Behavior**

```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TimeoutError: Clock stretch too long
```

**Description**

- Error while using i2c...
- Only happens when...
- might be related to #4291...

**Additional Info**

zestyping, 2022-01-28:
@tannewt Any insight into why this isn't working?  Do we know that the `rtc` module is even configured to read from the RTC chip in this case?

tannewt, 2022-01-31:
I haven't worked on this code in quite a while. We've never officially started supporting it with Adafruit-funded folks so it's unlikely we'll get to it soon.

It does look like the `rtc` module talks to the internal RTC. The place to look is here: https://github.com/adafruit/circuitpython/blob/main/ports/mimxrt10xx/common-hal/rtc/RTC.c

That's the code that configures the RTC of the device. `supervisor/port.c` may have some clock initialization in it as well. Losing the value may be a power config problem or re-init problem. I'm not sure which.

PR #7463 (2023-01-18): Use low power RTC on mimxrt10xx (Teensy41) boards
There are apparently two RTC interfaces in the mimxrt10xx dev kit. The low power interface access the battery backed up hardware.

I've tested this on the Teensy41 and it seems to fix issue #4574 (fixes #4574)
closed 2023-01-18 by tannewt