adafruit/circuitpython #4575: Feather M4, loses internal RTC values on reset, upgrade from 5.3.1 to 6.2.0

gertvb, 2021-04-08:
[code.py.txt](https://github.com/adafruit/circuitpython/files/6276872/code.py.txt)

**Firmware**

```python
Adafruit CircuitPython 6.2.0 "adafruit-circuitpython-feather_m4_express-en_GB-6.2.0.uf2"; Feather M4 Express
```

**Code/REPL**

```python

import rtc
import time
import supervisor

my_real_time_clock = rtc.RTC() #Uses Internal ADC on AtSAMD51

#To set the time one has to run this script with the following line uncommented.
#With the Lipoly battery connected to the Feather, the internal RTC keeps correct time, even when pressing the reset button
#This should only be needed to run once in the lifetime of the project
#my_real_time_clock.datetime = time.struct_time((2021, 04, 08, 08, 54, 05, 0, -1, -1))

while True:
    current_time = my_real_time_clock.datetime
    my_datestring = "{:2d}".format(current_time.tm_year) + "/{:02d}".format(current_time.tm_mon) + "/{:02d}".format(current_time.tm_mday)
    my_timestring = "{:02d}".format(current_time.tm_hour) +  ":{:02d}".format(current_time.tm_min) +     ":{:02d}".format(current_time.tm_sec)

    if supervisor.runtime.serial_connected:
        print ("Time : ", my_datestring, " ", my_timestring, "\n", sep = "")

    time.sleep(1)
```

**Behavior**

Code runs fine, except that the following now with Circuitpython 6.2.0 has to be set everytime the Feater is reset, whereas with Circuitpython 5.3.1 the RTC kept track of time even when the reset button was pressed

```python
my_real_time_clock.datetime = time.struct_time((2021, 04, 08, 08, 54, 05, 0, -1, -1))
```

**Description**

I am using the internal RTC on the AtSAMD51 to keep time in a project I am working on.

Under CircuitPython 5.3.1 the values stored in the internal RTC of the AtSAMD51  was not cleared when pressing the reset button

With CircuitPython 6.2.0 the internal RTC values are cleared when pressing reset

**Additional Info**

None

tgruetzm, 2022-05-26:
Any ideas on when this will be fixed? It's a deal breaker for a project I'm working on and hate to have to go to a M0 board...

dhalbert, 2022-05-27:
This code was changed here: https://github.com/adafruit/circuitpython/commit/7100d5e485a93552e035476cb794145d6e194052#diff-a944053c00651f658ae1cb445647c6509c09f656236817a32d11eded4ad449d1R143 about two years ago, before 5.4.0-beta.0. We added a `SWRST` of the `RTC`, in preparation for a rework of how ticks are used internally, so we could use less power during `time.sleep()`.

I am not sure we can undo that; we may depend on the tick registers starting at 0, and the RTC being in a known state. @tannewt do you know off the bat whether we could avoid the reset? We also SWRST the RTC in various cases involving `alarm` sleep states.

Can you use a battery-powered breakout board RTC, or are you too tight for space for that? I think we would recommend that as the best solution, since it will work through any disruption to the SAMD51 itself.

tgruetzm, 2022-05-27:
Thanks for the quick update.  I definitely could use an external RTC, it's just  a bummer that the board is advertised as including one.  Any sense of if this is SAMD51 specific?

tannewt, 2022-05-27:
I bet we could remove the SRST or do it conditionally on the state of the CTRL register.