adafruit/circuitpython #4593: Memory allocation fails fetching large json file

jedgarpark, 2021-04-12:
Running the [MagTag sports schedule viewer](https://learn.adafruit.com/magtag-sports-schedule-viewer/code-the-magtag-sports-schedule-viewer) code with the MLB schedule causes a memory allocation failure. This didn't rear its head until now because MLB was off season. Now the file is apparently too large and the MagTag crashes.

```File "code.py", line 247, in <module>
  File "code.py", line 189, in fetch_sports_data
MemoryError: memory allocation failed, allocating 103 bytes```

ladyada, 2021-04-12:
how big is the json? cause we have 2MB of PSRAM

jedgarpark, 2021-04-12:
http://site.api.espn.com/apis/site/v2/sports/baseball/mlb/scoreboard I just downloaded that and it is 328KB

todbot, 2021-04-12:
How does one determine object size in bytes in CircuitPython?  With regular Python one can do `sys.getsizeof()`. The best I could find is by inference of free memory:
```py
gc.collect()
before = gc.mem_free()
data = get_data_from_somewhere()
gc.collect()
after = gc.mem_free()
print( after - before )
```
Is there a better way?

tannewt, 2021-04-12:
@todbot I don't know of a better way from within CircuitPython. The heap analysis script I have could be useful. Maybe this is hitting a dictionary growth curve or something.

WarriorOfWire, 2021-04-13:
I noticed that the example code there uses
```
raw_data = json.loads(magtag.fetch(auto_refresh=False))
```

I added stream protocol support to json.load a while back to support memory-efficient large json document loads. Does magtag.fetch or some peer api return an object that implements circuitpython stream protocol?

usable like
`raw_data = json.load(magtag.fetch_stream(auto_refresh=False))` if it does. This would prevent a round trip through a string allocation saving 328+KB of continuous memory that is otherwise unusable during parsing.

I use this with json over uart like `response = json.load(board.uart)`

makermelissa, 2021-04-13:
I wrote the MagTag (and PortalBase) library and I wasn't aware of the stream protocol implementation, so there may be some room for improvement.

WarriorOfWire, 2021-04-13:
@makermelissa If I remember correctly a `bytes` will work but doesn’t avoid the large response allocation. Would want to push it all the way back to an object with a read(n) that gets the data from the network buffer e.g., some `response.body` type of thing. In that way you could deserialize the json as the bytes arrive, keeping your response buffer trim and a more polite memory footprint with small objects.

todbot, 2021-04-13:
Ooo a streaming parser!  I would very much be interested in testing that out.

I did some [quick tests of that 400kB ESPN JSON](https://gist.github.com/todbot/3d7b393e1b23095498d854fecfa901af) and it appears to occupy 1.2 MB of RAM (and 60 seconds) after parsing the entire thing into one big object. This matches my expectations of an in-memory tree parser. So I suspect JP's literally just hitting the RAM limit.

WarriorOfWire, 2021-04-14:
So need to pass `stream=True` in the requests.get invocation here https://gist.github.com/todbot/3d7b393e1b23095498d854fecfa901af#file-json_memtest-py-L22

and for that to actually work we’d need to update the adafruit_requests library to do something with it (e.g., not receiving the whole response from the socket in the method) here https://github.com/adafruit/Adafruit_CircuitPython_Requests/blob/c55425e17842cce3ec0b34489133436c3e1a3898/adafruit_requests.py#L529
The stream parameter is plumbed but not consumed in the session. Feel free to take a crack at it or I can at some point. Looks like @tannewt has been in that file and thought about this so maybe he has knowledge of what needs to happen.

In theory moving that recv_into into the socket when stream=True would be a first step.

tannewt, 2021-04-14:
Requests will stream when parsing JSON when you call `.json()` on the response object too.

WarriorOfWire, 2021-04-14:
Oh yeah sure enough if I read more closely it’s not eagerly filling the buffer. `requests.get().json()` appears to do the right thing and handles fancy stuff like chunked downloads too.

tannewt, 2021-07-15:
Fixed by the streaming parsing.

closed 2021-07-15 by tannewt

jepler, 2023-06-20:
The library for streaming json is https://github.com/adafruit/Adafruit_CircuitPython_JSON_Stream