adafruit/circuitpython #4596: toupad doesn't work in esp-s2

fredqian, 2021-04-12:
fireware: 6.2.0 release(any version after 6.0.1)
hardware: any esp32-s2 board 
Test code:
import time
import board
import touchio
Atouch4 = touchio.TouchIn(board.IO4 )
touch5 = touchio.TouchIn(board.IO5 )
touch6 = touchio.TouchIn(board.IO6 )
touch7 = touchio.TouchIn(board.IO7 )
while True:
    if touch4.value:
        print("touch4 Touched!")
    if touch5.value:
        print("touch5 Touched!")
    if touch6.value:
        print("touch6 Touched!")
    if touch7.value
        print("touch7 Touched!")
    time.sleep(0.5)

The Output is :
touch4 Touched!
touch5 Touched!
touch6 Touched!
touch4 Touched!
touch5 Touched!
touch6 Touched!
touch4 Touched!
touch5 Touched!
touch6 Touched!
touch4 Touched!
touch5 Touched!
touch6 Touched!
touch4 Touched!
touch5 Touched!
touch6 Touched!

error: Only the lastest touchpad can work!

moltar808, 2021-04-13:
You forgot the „:“ in the last if clause

dhalbert, 2021-04-13:
I'm not sure what you're saying? Are you saying `board.IO7` isn't working as a `TouchIn`, or something else? Are you touching 4, 5, 6, 7 in sequence.

I edited your initial post to put the code and output in triple backquotes: ` ``` ` so it's formatted nicely. Also I fixed the typos, including what @moltar808 noted.

fredqian, 2021-04-13:
> 发送日期：2021-04-13 20:50:08
> 抄送人：fredqian ***@***.***>,Author ***@***.***>
> 主 题：Re: [adafruit/circuitpython] toupad doesn't work in esp-s2 (#4596)

I didn't touch 5,6,7but it return value is touched  
If I changed the init order to 4,6,7,5,then 4,6,7 will return touched. so. only the lastouchpad can work.

dhalbert, 2021-04-13:
There is a calibration problem. They are all reading some value, but some think they are being touched. Try this sample, and you'll see that the raw values change:
```py
import time
import board
import touchio
touch4 = touchio.TouchIn(board.IO4 )
touch5 = touchio.TouchIn(board.IO5 )
touch6 = touchio.TouchIn(board.IO6 )
touch7 = touchio.TouchIn(board.IO7 )
while True:
    print("touch4", touch4.value, touch4.raw_value)
    print("touch5", touch5.value, touch5.raw_value)
    print("touch6", touch6.value, touch6.raw_value)
    print("touch7", touch7.value, touch7.raw_value)
    print()
    time.sleep(1)
 ```

dhalbert, 2021-04-17:
Reopening: there is still a problem we need to fix to make sure the calibration is such that they are not "on" at the start.

dhalbert, 2021-07-15:
See #4311 (which we closed).

closed 2021-07-15 by dhalbert