adafruit/circuitpython #4598: REPL becomes silent if aesio and frozen modules are included in a build for esp32s2

aziascreations, 2021-04-13:
**Firmware**

```python
Adafruit CircuitPython 7.0.0-alpha.1-78-g37887c42a-dirty on 2021-04-13; Saola 1 w/Wrover with ESP32S2
```

<br>

**Code/REPL**

```python
 import wifi
```

<br>

**Behavior**

Nothing appears in the REPL shell anymore.

However, it is still listening, as pressing <kbd>CTRL</kbd>+<kbd>D</kbd> makes the on-board LED go back from from white back to "flashing" green.

And Putty keeps the connection open and can reconnect to it perfectly fine, even if the REPL doesn't come back still.

<br>

**Description**

I was trying to add frozen modules and the `aesio` module to a customized firmware for an ESP32-S2 Saola 1 WROVER board, and I noticed that when I combined both, I was unable to properly load the `wifi` module.

As soon as I imported it, the REPL shell only listened but didn't output anything.<br>
The only way I could gain back access to the REPL was by restarting the device using the *RST* button.

The only modifications I did to existing files were in `ports/esp32s2/boards/espressif_saola_1_wrover/mpconfigboard.mk` where I added the following lines:
```
FROZEN_MPY_DIRS += $(TOP)/frozen/hard_data
CIRCUITPY_AESIO ?= 1
```

And the frozen module referenced is a single file located in `frozen/hard_data/data_to_parse/data_to_hash.py` which contains the following text:
```python
data = """
Lyrics to the "Mama Papa Brickolini" song
"""
```

I tried compiling the firmware with and without the `aesio` module and the frozen module, and I noticed than when both of them where enabled, the REPL has trouble with the `wifi` module.

The same behavior was observed on the commit [e6aaf9924f6cf49c7ea8327f181960ecb2fd636c](https://github.com/adafruit/circuitpython/commit/e6aaf9924f6cf49c7ea8327f181960ecb2fd636c) which is another one I was playing around with before encountering this issue.

Other than that, the rest of the firmware compiled and was flashed perfectly without any errors, it is just this module that seems to have problems with it.<br>
Even the `aesio` and other frozen modules I tried worked perfectly.

tannewt, 2021-04-13:
CTRL-D is checked in a different task from the regular VM. I suspect the regular VM task is stuck somewhere.

dhalbert, 2021-04-13:
This is somewhat reminiscent to me of the I2C/wifi issue, but in that case, the USB connection was broken, not just silent.

aziascreations, 2021-04-14:
@tannewt I'm not familiar enough with the internal structure of CircuitPython to help with that, but I could try to re-compile the firmware with the `DEBUG=1` argument if that could help you or someone else.

<br>

I have also done a couple more tests and this issue seems to be also present on the official 6.2.0 release.
However, it seems like it is intermittent.

For the new tests, I tried reflashing the official firmware provided on circuitpython.org, and the REPL also froze when the `wifi` module was imported.
Thinking the internal flash could be faulty since I had re-flashed it quite a few times and played around with the dualbank previously, I reflashed a completely new board I hadn't touched once before, and the same problem occurred again.

This time it only occurred once, and after mashing the CTRL+D keys, it seems like it rebooted* and has been working properly after that point**.

*: Putty lost the connection and explorer re-opened the USB drive.
**: Haven't done to many tests yet, as I'm focused on another library.

tannewt, 2021-04-14:
I think the next step would be loading a DEBUG=1 build and capturing the debug UART output available on native USB of the saola. Thanks for looking into this!

aziascreations, 2021-04-16:
I'll look into that tonight then,and once I get something I'll post it here.

In the meantime, I tried the same code in the *code.py* file and also noticed the same behavior, except that I could not get back any kind of control with CTRL+C or CTRL+D.
And after leaving it for a 5-15 seconds, the onboard LED seems to be displaying an error code:

When importing the `wifi` module:
1x  Long Green
1x  Long White
1x  Short Pause
2x  Short Yellow
1x  Short Pause
9x Short Blue

When connection to an AP (Not seen as being connected on the AP)
1x  Long Green
1x  Long White
1x  Short Pause
2x  Short Yellow
1x  Short Pause
10x Short Blue

When creating or using a socket gotten via `wifi.radio` and `socketpool` (Data returned seemed to be the same in sequential NTP packets, not 100% sure)
1x  Long Green
1x  Long Yellow
1x  Short Pause
4x  Short Yellow
1x  Short Pause
8x Short Blue

And, in the event it can help, the USB drive was still accessible during all of this.