adafruit/circuitpython #4601: NeoPixel LEDs remain lit up from code when entering the REPL

kattni, 2021-04-13:
As it is, with the Neo Trinkey specifically, the non-status NeoPixels remain lit up if they were lit up from code in code.py when you enter the REPL. The status LED changes to indicate that you have entered the REPL. So, for example, if you had all four LEDs turned red in your code, and then entered the REPL, three would remain red, and one would turn white. This confused me as I moved onto something else and forgot I had entered the REPL - I figured out what was going on when I came back to it and found I was in the REPL (I thought something had become corrupted). I imagine this might confuse others as well.

Is it worth turning off the remaining non-status LEDs when entering the REPL to avoid confusion? This is only relevant on boards with more than one LED, where one of those LEDs only is a status LED.

@ladyada How would you want this to work?

This is what it looks like when the LEDs were red from code, and then you enter the REPL:
[image]

ladyada, 2021-04-13:
we should clear on soft rest - @dhalbert would know if theres' any hook for running a neopixel_write(0) during a soft reboot

tannewt, 2021-04-13:
There is a `reset_board function`. Here is the CPX version: https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/boards/circuitplayground_express/board.c#L55

ajs256, 2021-04-25:
Done in my fork: https://github.com/ajs256/circuitpython/commit/4bafbca0c3571af11163b491994a3a8a7c24ea03

I will open a PR if CI goes green.

kattni, 2021-04-26:
@ajs256 GitHub recently changed it so that first-time contributors PRs require approval to run CI. This was put in place to avoid a recent spate of CI abuse. Thank you for your patience with the system.

PR #4667 (2021-04-26): Trinkey pixel fixes
Add a `board_reset_user_neopixels` to the Slidey Trinkey, NeoTrinkey, and Prox Trinkey `board.c` files to reset the NeoPixels when entering REPL.

**This is untested as I do not have hardware.**

Closes #4601.
closed 2021-04-26 by kattni