adafruit/circuitpython #4602: pulseio.PulseOut timing very wrong on CPX (SAMD21) on 6.x - breaks infrared sending

kevinjwalters, 2021-04-13:
I noticed my CPX was still on `5.3.1`. When I upgraded it to `6.2.0` the IR sending stopped working, I could still see the IR flashing with the aid of a camera. Checked `6.0.0` and that's also broken.

I tried to pick up the signal at the IR diode with my (digital) SQ25 analyser but a few things conspired against me there. I duplicated the sending code and sent pulses to `A2` too. That shows timing of the pulses is good on `5.3.1` and is all over the place on `6.2.0`.

### Good 5.3.1

2400, 600, 1200, 600, 600, 600 us is in the program and trace shows these values:

[image]

### Bad 6.2.0

[image]

Trace shows 2400, 944, 852, 798, 402, 899, only first value is good.

I've not looked closely at all of the trace but perhaps the falling edge is good and the timing of rising edge is kaput?

tannewt, 2021-04-14:
Please try 6.0.0 as well. I think that's when we switched time keeping internally.

kevinjwalters, 2021-04-14:
`6.0.0` is broken too

dhalbert, 2021-04-14:
This reminds me of #3626 too.

kevinjwalters, 2021-04-14:
The underlying PWM 20% 40kHz carrier is fine, btw.

DavePutz, 2021-04-15:
@kevinjwalters - In my tests I'm seeing that the pulses are accurate after I power cycle the CircuitPlayground. They stay accurate until I change the duty cycle; then they stay messed up until I power cycle the board again. Can you verify whether or not you see the same thing? If so, there may be something not getting reset properly when we set up the PulseOut.

kevinjwalters, 2021-04-15:
I'm not fiddlng with `duty_cycle` just setting it once in constructor.

I just noticed code can misbehave in a way I've not seen before on `6.2.0`. The output can stop showing up on serial console and control-c doesn't work but `CIRCUITPY` is accessible and still works. The code can also get stuck but `CIRCUITPY` still ok, that's weird combination.

DavePutz, 2021-04-18:
Here is a simplified example that demonstrates the issue:
`
from adafruit_irremote import  GenericTransmit
import pulseio
from board import A2
from time import sleep

# 20% duty cycle at 40kHz.
pwm = pulseio.PWMOut(A2, frequency=40000, duty_cycle=13107)
pulse = pulseio.PulseOut(pwm)
ir_encoder = GenericTransmit(header=[2400, 600],
                                               one=   [1200, 600],
                                               zero=  [600,  600],
                                               trail=None)
ir_encoder.transmit(pulse, [0xB4, 0xB8, 0xF1],
                    repeat=2, delay=0.005, nbits=24)
pulse.deinit()
pwm.deinit()
`
If this code is run immediately after a hard reset it works correctly. Instrumentation shows that the timer count register is reliably 2us higher than the compare value. However, if anything is done via USB (i.e. changing a file in CIRCUITPY) then the difference at interrupt time between the timer compare value and the timer count register varies widely; from 2us to  250us. This will continue until another hard reset occurs (soft reset does not change things).
Still investigating what USB activity has to do with interfering with proper timing to get to the interrupt handler.

kevinjwalters, 2021-04-18:
@DavePutz Is this some sort of deferred processing due to incorrect interrupt masking then? BTW, when you say anything over USB does that include connecting to serial console? That's an easy test to differentiate between file system stuff and all USB.

DavePutz, 2021-04-18:
@kevinjwalters - sorry I should have been more specific. The USB console
does not seem to have any impact. I
get the incorrect timings after I do anything that writes to the
CIRCUITPY drive (reads seem to be OK)

DavePutz, 2021-04-20:
@kevinjwalters - Can you build from my branch at https://github.com/DavePutz/circuitpython/tree/issue_4602 and see if the fix I put in takes care of your issue? Please let me know; if your test is OK I'll open a PR.

DavePutz, 2021-04-23:
I was able to test on a Sony camera and firing the shutter worker correctly. Opening a PR.

DavePutz, 2021-04-26:
Fix is in pull request #4655.

tannewt, 2021-07-28:
Fixed by #4655

closed 2021-07-28 by tannewt