adafruit/circuitpython #4610: Sparkfun Pro Micro RP2040 GPIO23 not mapped to CO pin

aerialviews007, 2021-04-14:
**Firmware**

6.2.0

```python
Adafruit CircuitPython 6.2.0-beta.2 on 2021-03-01; Raspberry Pi Pico with rp2040
```

**Code/REPL**

keyboard.row_pins = (board.MOSI, board.A0, board.A1, board.A2, board.A3)

or

keyboard.row_pins = (board.GP23, board.A0, board.A1, board.A2, board.A3)

**Behavior**

AttributeError: 'module' object has no attribute 'MOSI'

**Description**

I'm attempting to use the RP2040 Pro Micro in a kmk keyboard application where a traditional pro micro would be used.  The pin needed is Pin 16 aka B2 and also MOSI on the Sparkfun pro micro.  This pin is labelled "CO" on the RP2040 board and per the sparkfun documentation, maps to GPIO23.  However, this does not work.  Reaching out to sparkfun, they say it definitely is GPIO23 and that it must be software.

**Additional Info**

ladyada, 2021-04-15:
hmm - it looks like its here ... 
https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/boards/sparkfun_pro_micro_rp2040/pins.c#L15
maybe you've got the wrong firmware installed? or a typo?

aerialviews007, 2021-04-15:
No, firmware is correct.  MOSI, CO and GP23 do not work.

dhalbert, 2021-04-15:
Download from here: https://circuitpython.org/board/sparkfun_pro_micro_rp2040/

closed 2021-04-15 by ladyada