adafruit/circuitpython #4615: Implement central for _bleio HCI

dhalbert, 2021-04-15:
Currently _bleio HCI support is for peripheral only.

shulltronics, 2022-06-20:
Hey there,

Is anyone currently working of or thinking about this enhancement? I'd like this feature for a BLE to Ethernet gateway I'm working on.

If there is anything important I should know about the current status of this feature before I dive in, please let me know. Thanks! :)

dhalbert, 2022-06-20:
Thanks for looking at this! There is commented-out code from the nRF52840 implementation that is in the base routines needed.

I don't remember if there is substantial extra stuff that needs to be implemented to do the bookkeeping necessary for HCI. For peripheral mode, I had to write a lot of new code, which is mostly in `att.c`. The HCI interface is rather low-level in many ways, and assumes a "smart" caller that handles a lot of the state.

shulltronics, 2022-06-20:
This will definitely be a test of my firmware skills, but I'm willing to try! :) I'm wrapping my head around the circuitpython structure and going to be focusing at first on getting it to compile for a CLUE board and also for an RP2040 (which will be communicating with an AirLift Featherwing for BLE).

Would you mind commenting on the following things for me?
1. Could you link more specifically to the commented code you're referring to for the nRF52840?
2. I've noticed that on my CLUE board, I can use BLE central mode. Does this mean that the central role is unimplemented only when using BLE via an HCI adapter?
3. Am I correct in thinking that this is specific to circuitpython, i.e. the `nina-fw` has all of the capabilities needed to act as BLE central?

Thanks!

dhalbert, 2022-06-20:
The commented-out code is in https://github.com/adafruit/circuitpython/tree/main/devices/ble_hci/common-hal/_bleio, in various files, most notably `att.c` and `Connection.c`, but may also be elsewhere.

The nRF52840 implements both peripheral and central mode just fine. If you can use an nRF52840, such as on the CLUE, or an nRF52840 Feather, then it will save a lot of work here. We won't hold it against you if you want to go this way :smile:  .

NINA-FW should have what you need. HCI is more or less a standard, so anything implementing HCI can work. HCI is not as high-level as one might like, as I mentioned.

Given that we now have ESP32-nn support, some of which do BLE, this issue is rather low priority. We are working on ESP32-S2 `_bleio` support, but may require some rework of some of the middleware (specifically `nimble`) to be as flexible as `_bleio`, which was originally done for the nRF52840. That work may be even harder.