adafruit/circuitpython #4621: Infinite hang upon creating I2CDevice

ajs256, 2021-04-17:
**Firmware**

```python
Adafruit CircuitPython 6.2.0 on 2021-04-05; Raspberry Pi Pico with rp2040
```

**Code/REPL**

```python
import board
import busio
import bitbangio
from adafruit_bus_device.i2c_device import I2CDevice

i2c = busio.I2C(board.GP17, board.GP16)
# i2c = bitbangio.I2C(board.GP17, board.GP16)

addr = 0x40

while not i2c.try_lock():
    print("i2c not locked, retrying...")

print("i2c initted")

dev = I2CDevice(i2c, addr)  # if we try to probe, it infinitely hangs
print("i2c device created")

```

**Behavior**

It infinitely hangs unless I pass in `probe=False` to the `I2CDevice` call.

**Description**

I have confirmed this with a [SparkFun SerLCD (ATmega328-based)](https://github.com/sparkfun/OpenLCD/) and an [HTU21D breakout](https://www.sparkfun.com/products/retired/12064). It seems to happen both with `busio` and `bitbangio`.

dhalbert, 2021-04-17:
Are both devices physically connected when you are trying this probe, or are you connecting one at a time?

ajs256, 2021-04-17:
One at a time.

ajs256, 2021-04-18:
And... just when I thought I was done with them, I am getting `OSError: [Errno 19] Unsupported operation` on i2c writes. Here is the code I am using: [gist link](https://gist.github.com/ajs256/571c67c654b560cd6fddb30c6dd3aa01)
Here's the traceback:
```python
Traceback (most recent call last):
  File "code.py", line 25, in <module>
  File "adafruit_bus_device/i2c_device.py", line 84, in write
OSError: [Errno 19] Unsupported operation
```
It rarely happens, so I can't make a minimal example.

Neradoc, 2021-04-26:
> It infinitely hangs unless I pass in `probe=False` to the `I2CDevice` call.

I believe that's because you lock the device with your `try_lock()` loop.
If you do that, use the `I2C` object directly.
The role of `I2CDevice` is to manage the locks. You use it like this:
```py
with dev as bus_device:
    bus_device.write(...)
```
(Note that here `bus_device == dev`, but I'd call it an implementation detail I think)

Now on the other issue, I do encounter random `OSError: [Errno 19] Unsupported operation` with the SerLCD. It tends to happen cumulatively after a series of writes, but seems to go away if I add a 1 ms sleep between each write.

[Here is my test code (running on QT Py 2040 + SparkFun 16x2 SerLCD - RGB Backlight)](https://gist.github.com/Neradoc/dbbbf8890f6f9dccf198b8b8127c18f9)

ajs256, 2021-05-04:
Thank you so much @Neradoc, those fixes worked!

Leaving this open so someone can figure out why the delay is needed at all.