adafruit/circuitpython #4630: Feather M0 Adalogger mpconfigboard.h has SDA and SCL swapped relative to schematic and pins.c

anecdata, 2021-04-18:
This arose from an issue on [Discord](https://discord.com/channels/327254708534116352/537365702651150357/833124505504317451). I don't have this board to try it out.

**Firmware**

```python
Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit Feather M0 Adalogger with samd21g18
```

**Code/REPL**

In https://github.com/adafruit/Adafruit_CircuitPython_ServoKit library:
```python
i2c = board.I2C()
```

**Behavior**

```python
code.py output:
Traceback (most recent call last):
  File "code.py", line 10, in <module>
  File "adafruit_servokit.py", line 79, in __init__
ValueError: Invalid pins
```

**Description**

[mpconfigboard.h](https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/boards/feather_m0_adalogger/mpconfigboard.h)  has SCL on PA22, SDA on PA23 for `DEFAULT_I2C_BUS_`
[pins.c](https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/boards/feather_m0_adalogger/pins.c) and [schematic](https://learn.adafruit.com/adafruit-feather-m0-adalogger/downloads) have SDA on PA22, SCL on PA23)

ladyada, 2021-04-18:
w00pz. wanna submit a fix? not a lot of folks used the adalogger :)

anecdata, 2021-04-18:
sure, I just have no way to test it

anecdata, 2021-04-18:
Fixed by #4631. Closing.

closed 2021-04-18 by anecdata