adafruit/circuitpython #4632: Pushes to temporary patch branches get uploaded to S3

Neradoc, 2021-04-18:
On occasion a patch branch on the adafruit/circuitpython repository is created for a Pull Request. This commit will be uploaded to S3, since every commit is uploaded to S3 regardless of the branch and whether or not it comes from an accepted PR. I feel like that could be confusing when pointing people to S3 to try "the latest".
Could/should those be skipped for the S3 upload in CI ?

tannewt, 2021-04-19:
Ya, I think we should only include pushes to main. I think this is what to change: https://github.com/adafruit/circuitpython/blob/main/.github/workflows/build.yml#L403

Neradoc, 2021-05-10:
I think changing all the relevant ifs to add `github.ref == 'refs/heads/main'` would work to limit pushes without interfering with releases (on 6.x branches for example), but I don't know how to test it:
> if: (github.event_name == 'push' && github.ref == 'refs/heads/main') || (github.event_name == 'release' && (github.event.action == 'published' || github.event.action == 'rerequested'))

I followed the doc example of a push action that is limited to the main branch:
https://docs.github.com/en/actions/reference/context-and-expression-syntax-for-github-actions#determining-when-to-use-contexts
> In this example, the if statement checks the github.ref context to determine the current branch name; if the name is refs/heads/main, then the subsequent steps are executed.

tannewt, 2021-05-11:
I think you can test on your own fork with its branches. The S3 upload steps will be attempted but fail (I think.) We can also just do best guess and check to see how it changes for this repo.

PR #5027 (2021-07-21): Only push `main` builds to S3
That will ensure it is always the latest instead of a .x or feature
branch.

Fixes #4632

Not tested but is copy and paste of existing stubs logic. Will check after merge.
commit ec6e432f (2021-07-21): Only push `main` builds to S3

That will ensure it is always the latest instead of a .x or feature
branch.

Fixes #4632
closed 2021-07-21 by microDev1