adafruit/circuitpython #4639: Unable to send HIDService keyboard events to specific connections

ATMakersBill, 2021-04-20:
**Firmware**
Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit Circuit Playground Bluefruit with nRF52840

**Behavior**
This code allows two bluetooth devices (iPhone, iPad, Android, PC, etc.) to connect as a central and bond to the device.

When "kl1.write(c)" is called on  line 71, the keystroke is sent to BOTH devices, not just the one connected to kbd1.

It seems there is no way to associate a Service (HIDService in this case) with just one of the connections - nor is it possible to tell the service which connection to send the HID events to.

This type of setup where the CP device allows multiple BLE Centrals to connect and receive keyboard events is critical to Assistive Technology users (like @cyborg55, @milador and many others).  The users want to be able to pair a phone, iPad, speech generating device (SGD) and PC to the same device that allows them to trigger actions via AT Switches.  Connections can be dropped and picked up, but we must be able to send keystrokes to a specific connection and keep track of which connection we're talking to.

While there may be a way to do this in CP, I've been working with @dhalbert, @David.G and others and have been stuck here for a while.

@dhalbert suggests that the issue might be at [this section of Characteristic.c](https://github.com/adafruit/circuitpython/blob/18548fc5d252f95a7304af8df09610c1160ce388/ports/nrf/common-hal/_bleio/Characteristic.c#L156)

I am good to find another solution, but I'm at a loss as it seems that Characteristics and Services seem detached from the Connections.  
**Code/REPL**

```python
#Two keyboard test

import adafruit_ble

from adafruit_ble.advertising import Advertisement
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.standard.hid import HIDService
from adafruit_ble.services.standard.device_info import DeviceInfoService
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS

import time
import board
import sys

hid1 = HIDService()
device_info = DeviceInfoService(software_revision=adafruit_ble.__version__, manufacturer="Adafruit Industries")
advertisement1 = ProvideServicesAdvertisement(hid1)
advertisement1.appearance = 961
scan_response1 = Advertisement()
scan_response1.complete_name = "CircuitPython HID1"

ble = adafruit_ble.BLERadio()

if not ble.connected:
    print("advertising1")
    ble.start_advertising(advertisement1, scan_response1)
else:
    print("already connected")
    print(ble.connections)

# This is a BLE keyboard
kbd1 = Keyboard(hid1.devices)

hid2 = HIDService()
advertisement2 = ProvideServicesAdvertisement(hid2)
advertisement2.appearance = 961
scan_response2 = Advertisement()
scan_response2.complete_name = "CircuitPython HID2"

# If I do the following before I get a connection on the first advertisement...
#   ble.start_advertising(advertisement2, scan_response2)
# It produce the following error...
#   _bleio.BluetoothError: Already advertising.

# This is a trick to wait for a first connection before making a new advertisement
while len(ble.connections) < 1:
    pass

print("advertising2")
ble.start_advertising(advertisement2, scan_response2)

# This is a BLE keyboard
kbd2 = Keyboard(hid2.devices)

kl1 = KeyboardLayoutUS(kbd1)
kl2 = KeyboardLayoutUS(kbd2)

m=5
s=1

# This works even if mouse2 is not connected yet
while True:
    while not ble.connected:
        pass
    print("Sending Keyboard input to just the first paired device")

    while ble.connected:
        c = sys.stdin.read(1)
        sys.stdout.write(c)
        kl1.write(c)
        #kl2.write(c)

# This is far from perfect, we only advertise 1 where the second is not there too.
    ble.start_advertising(advertisement1)
```

**Additional Info**
I have been testing with a BLE CircuitPlayground.  It successfully lets you connect multiple devices and then sends keystrokes to both (even when you only request it on one of the HIDServices).

tannewt, 2021-04-20:
BLE GATT doesn't really expose the idea of per-connection interaction but I believe the soft device does allow for it. Are the HID reports actually sent out via notify? PacketBuffer may be a better start because it will only talk to the first connection to write or subscribe to it.

dhalbert, 2021-04-20:
Yes, they are via notify, e.g. https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/df40a241841918d6c0a984d427e61ca831b7b431/adafruit_ble/services/standard/hid.py#L172.

I thought this might have to do with not handling duplicate services correctly. We need to see what the handles are for the duplicate characteristics. I looked again (briefly) at the SD API and it seems to handle this case, as you said.

ThomasAtBBTF, 2022-01-14:
I am "suffering" from the same problem as described initially by ATMakesBill.
We are trying to make a BrailleKeyboard.
While investigating the problem and trying to find a solution I think there are two very different ways to achieve what we want:
1. Checking the connections and disconnecting the "undesired" centrals. This would require a API which is able to identify the central at the "other side" of the connection.
2. Changing the MAC Address of the peripheral to different  values and connect / pair / bond each desired central to "its" MAC-Address. (This is how a Seenda Bluetooth Keyboard with a Nordic Chip is handeling it's multi pairing feature)

But for both options, there is currently no API available.

tannewt, 2022-01-18:
Do you need a unique id or would model name be enough. With the existing API you should be able to access a central Device Information Service to read it's model name. The spec for the service is here: https://www.bluetooth.com/specifications/specs/device-information-service-1-1/

dhalbert, 2022-01-18:
Here is an example of a central using Device Information Service to find out about a peripheral. But it could work the other way as well: https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/99a7e2d70aaf87b14064917d71de97d3e5a753c4/CLUE_Cycling_Simple/code.py